motioneye-project/motioneye #1: motioneye.py dated 8/30/2015 is 0 bytes

drknstrmyknight, 2015-08-30:


ccrisan, 2015-08-30:
Yes, the file is no longer used and will be removed. The migration to setuptools and github is not yet complete. I'm working on some wiki pages that will help users install, configure and run motionEye on their systems. Until then, please stick with the bitbucket 0.26 version.

closed 2015-08-30 by ccrisan

drknstrmyknight, 2015-08-30:
Well, that was a wasted morning...

ccrisan, 2015-08-30:
I'm sorry, what exactly was a wasted morning? For whom? I'm not sure I understand the issue here.

drknstrmyknight, 2015-08-30:
going through the motions of installing 27.1Don't get me wrong, you're going great work, I just got caught in the flux

ccrisan, 2015-08-30:
This migration to github, to PyPI and everything is long overdue. I knew it wouldn't be easy but it had to be done. As soon as I manage to put together new install docs, things will become a lot clearer for everyone.

drknstrmyknight, 2015-09-01:
Had to change my source to squeeze to get ffmpeg installed on debian jessie - all is good now.

ccrisan, 2015-09-01:
That's odd. I tried it with debian 8 and it went alright with "jessie". What was the problem, so I can update the docs?

drknstrmyknight, 2015-09-18:
From: "frank@boffey.com" <frank@boffey.com>
 To: ccrisan/motioneye <reply@reply.github.com> 
 Sent: Tuesday, September 1, 2015 7:07 AM
 Subject: Re: [motioneye] motioneye.py dated 8/30/2015 is 0 bytes (#1)
   
Had to change my source to squeeze to get ffmpeg installed on debian jessie - all is good now.