motioneye-project/motioneye #2: Ip Camera broken pipe

jacopomarcovaldi, 2015-08-31:
Hi,
i have a problem with motioneye and my nsc18-wn ipcam.
This ip camera has a url for mjpeg stream such as http://[ip]/loginfree.jpg.
I use this url in motion eye and for some time it works but toda i get this error in log file: 

[0] Processing thread 0 - config file /home/pi/motioneye/conf/motion.conf
[0] Processing config file thread-1.conf
[0] Unknown config option "locate_motion_style"
[0] Processing config file thread-2.conf
[0] Unknown config option "locate_motion_style"
[0] Motion 3.2.12 Started
[0] ffmpeg LIBAVCODEC_BUILD 3482368 LIBAVFORMAT_BUILD 3478785
[0] Thread 1 is from thread-1.conf
[0] Thread 2 is from thread-2.conf
[1] Thread 1 started
[1] cap.driver: "uvcvideo"
[1] cap.card: "Trust Webcam"
[2] Thread 2 started
[0] motion-httpd/3.2.12 running, accepting connections
[0] motion-httpd: waiting for data on port TCP 7999
[1] cap.bus_info: "usb-20980000.usb-1.2"
[1] cap.capabilities=0x84200001
[1] - VIDEO_CAPTURE
[1] - STREAMING
[1] Config palette index 8 (YU12) doesn't work.
[1] Supported palettes:
[1] 0: YUYV (YUV 4:2:2 (YUYV))
[1] Selected palette YUYV
[1] Test palette YUYV (352x288)
[1] Using palette YUYV (352x288) bytesperlines 704 sizeimage 202752 colorspace 00000008
[1] found control 0x00980900, "Brightness", range -64,64
[1]     "Brightness", default -8193, current 33
[1] found control 0x00980901, "Contrast", range 0,64
[1]     "Contrast", default 57343, current 32
[1] found control 0x00980902, "Saturation", range 1,128
[1]     "Saturation", default 57343, current 64
[1] found control 0x00980903, "Hue", range -40,40
[1]     "Hue", default -8193, current 0
[1] found control 0x00980910, "Gamma", range 72,500
[1]     "Gamma", default 57343, current 100
[1] found control 0x00980913, "Gain", range 0,100
[1]     "Gain", default 57343, current 0
[1] mmap information:
[1] frames=4
[1] 0 length=202752
[1] 1 length=202752
[1] 2 length=202752
[1] 3 length=202752
[1] Using V4L2
[1] Resizing pre_capture buffer to 1 items
[2] Info: No 'Connection: Keep-Alive' nor 'Connection: close' header received. Motion continues unchanged.
[2] Resizing pre_capture buffer to 1 items
[2] Error sending 'connect' request: Broken pipe
[2] Error sending 'connect' request: Broken pipe
(errore repeated for a while)
[2] Started stream webcam server in port 10011
[2] Resizing pre_capture buffer to 7 items
and then again
[2] Error sending 'connect' request: Broken pipe
[2] Error sending 'connect' request: Broken pipe

Here is my configuration file for this camera (i have also a usb camera attached but it works fine)

# @working_schedule_type outside
# @webcam_resolution 100
# @network_password
# @enabled on
# @name Ip_Camera1
# @network_server
# @id 2
# @motion_detection off
# @network_username
# @storage_device custom-path
# @preserve_movies 0
# @network_share_name
# @preserve_pictures 0
# @working_schedule
# @webcam_server_resize off

snapshot_interval 0
minimum_motion_frames 5
webcam_maxrate 5
target_dir [my-DIR]
text_double on
webcam_quality 85
threshold 1999
quality 85
text_left Ip_camera1
post_capture 4
noise_level 31
framerate 2
netcam_url http:// [IP-CAMERA] /loginfree.jpg
pre_capture 2
noise_tune on
webcam_motion off
locate_motion_style redbox
jpeg_filename %m-%d-%Y-%H-%M-%S
locate on
webcam_port [PORT]
ffmpeg_variable_bitrate 0
output_all off
ffmpeg_video_codec msmpeg4
movie_filename %d-%m-%Y/%H-%M-%S
ffmpeg_cap_new on
auto_brightness on
text_changes on
on_event_end /home/pi/motioneye/eventrelay.py stop %t
rotate 0
text_right %d-%m-%Y\n%T
on_event_start /home/pi/motioneye/eventrelay.py start %t; /home/pi/motioneye/sendmail.py "out.mail.it" "587" "email@email.com" "password" "False" "email@sendto.com" "motion_start" "%t" "%Y-%m-%dT%H:%M:%S" "8"
ffmpeg_bps 43008
lightswitch 50
output_normal on
gap 10
webcam_localhost off
snapshot_filename
max_mpeg_time 0
on_movie_end sh /motionwhatsappimg.sh %f [phone_number]; sh /motionwhatsapp.sh

netcam_http 1.1
netcam_tolerant_check on

Can you help me?
Thanks.
Jacopo

ccrisan, 2015-09-01:
Well, I cannot tell too much from your log file apart from the fact that there's a network connectivity issue. Make sure your network connection works correctly. Have you upgraded your netcam's firmware recently?

P.S. the next version will contain many netcam-related fixes and improvements. If your problem persists, it may go away with the next version.

jacopomarcovaldi, 2015-09-01:
Thank you for your quick reply.
Well, yes i upgraded netcam's firmware.
Strange thing is that when i add in motioneye my netcam it works.
When i try to change a setting from gui and i press apply camer window become green and the grey and the message is "connection lost..".
I am pretty sure that it's not e network issue, infact when i browse "http:// [IP-CAMERA] /loginfree.jpg" with a browser it works.
Jacopo

ccrisan, 2015-09-01:
If the trouble starts *after the first apply* of settings, then it's a bug in motionEye that has been fixed recently.

jacopomarcovaldi, 2015-09-01:
ok great :)
I hope to be able to download new version soon :)
Thank you for your work.

closed 2015-09-01 by jacopomarcovaldi