motioneye-project/motioneye #3: failed to get motion detection status for camera with id

francesconuma, 2015-08-31:
Any hint?

2015-08-31 18:12:10: [motioneye]     INFO: hello! this is motionEye server 0.27.1
2015-08-31 18:12:15: [motioneye]     INFO: cleanup started
2015-08-31 18:12:15: [motioneye]     INFO: wsswitch started
2015-08-31 18:12:15: [motioneye]     INFO: thumbnailer started
2015-08-31 18:12:15: [motioneye]     INFO: server started
2015-08-31 18:12:25: [motioneye]    ERROR: failed to get motion detection status for camera with id 1: HTTP 599: Failed to connect to 127.0.0.1: Network is unreachable
2015-08-31 18:12:25: [motioneye]  WARNING: skipping motion detection status update for camera with id 1
2015-08-31 18:12:35: [motioneye]    ERROR: failed to get motion detection status for camera with id 1: HTTP 599: Failed to connect to 127.0.0.1: Network is unreachable
2015-08-31 18:12:35: [motioneye]  WARNING: skipping motion detection status update for camera with id 1
...
2015-08-31 18:15:05: [motioneye]    ERROR: failed to get motion detection status for camera with id 1: HTTP 599: Connection timed out after 5041 milliseconds
2015-08-31 18:15:05: [motioneye]  WARNING: skipping motion detection status update for camera with id 1
2015-08-31 18:15:15: [motioneye]    ERROR: failed to get motion detection status for camera with id 1: HTTP 599: Connection timed out after 5041 milliseconds
2015-08-31 18:15:15: [motioneye]  WARNING: skipping motion detection status update for camera with id 1
2015-08-31 18:15:17: [motioneye]  WARNING: control hue not found for device /dev/video0
2015-08-31 18:15:18: [motioneye]  WARNING: control hue not found for device /dev/video0
2015-08-31 18:15:19: [motioneye]  WARNING: control hue not found for device /dev/video0
2015-08-31 18:15:25: [motioneye]    ERROR: failed to get motion detection status for camera with id 1: HTTP 599: Connection timed out after 5041 milliseconds
....

ccrisan, 2015-09-01:
Have you re-added your camera after upgrading to 0.27? Or is this a completely new installation?

francesconuma, 2015-09-01:
I removed the older version (folder and rc.local script) and installed the 0.27 as explained in the new wiki.
Also tried removing and reinstalling motion via apt-get (raspbian).
I also launched apt-get update /upgrade and firmware update on raspberry.
Maybe I can try from a new fresh image.
Regards.

francesconuma, 2015-09-01:
A fresh install resolved the problem.
Now It's all working.

closed 2015-09-01 by francesconuma