motioneye-project/motioneye #4: usb webcam no more working

sgiannulli, 2015-09-01:
Hi Calin,
I'm on RPi with Raspbian; installed new 0.27 version and facing the following problems:
a usb webcam that perfectly worked with 0.26 version now does not work anymore;
motion.log:
[1] Config palette index 8 (YU12) doesn't work.
[1] Supported palettes:
[1] 0: YUYV (YUV 4:2:2 (YUYV))
[1] 1: MJPG (MJPEG)
[1] Selected palette YUYV
[1] VIDIOC_TRY_FMT failed for format YUYV: Input/output error
[1] Unable to find a compatible palette format.
[1] ioctl (VIDIOCGCAP): Inappropriate ioctl for device
[1] Retrying until successful connection with camera
[1] cap.driver: "uvcvideo"
[1] cap.card: "UVC Camera (046d:08cf)"
[1] cap.bus_info: "usb-20980000.usb-1.3.1"
[1] cap.capabilities=0x84200001
[1] - VIDEO_CAPTURE
[1] - STREAMING
motioneye.log:
2015-09-01 18:31:51: [motioneye]  WARNING: control hue not found for device /dev/v4l/by-id/usb-046d_08cf-video-index0
2015-09-01 18:31:51: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/7'
2015-09-01 18:31:51: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/6'
2015-09-01 18:31:51: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/5'
2015-09-01 18:31:51: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/4'
2015-09-01 18:31:51: [motioneye]  WARNING: control hue not found for device /dev/video1
2015-09-01 18:31:51: [motioneye]    ERROR: Exception in callback None
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado/ioloop.py", line 865, in start
    fd_obj, handler_func = self._handlers[fd]
KeyError: 5
2015-09-01 18:31:51: [motioneye]    ERROR: Exception in callback None
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado/ioloop.py", line 865, in start
    fd_obj, handler_func = self._handlers[fd]
KeyError: 12
2015-09-01 18:31:51: [motioneye]    ERROR: Exception in callback None
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado/ioloop.py", line 865, in start
    fd_obj, handler_func = self._handlers[fd]
.........................................
2015-09-01 19:49:24: [motioneye]     INFO: hello! this is motionEye server 0.27.1
2015-09-01 19:49:29: [motioneye]     INFO: cleanup started
2015-09-01 19:49:29: [motioneye]     INFO: wsswitch started
2015-09-01 19:49:29: [motioneye]     INFO: thumbnailer started
2015-09-01 19:49:29: [motioneye]     INFO: server started
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/32'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/31'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/28'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/27'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/26'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/25'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/24'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/23'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/22'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/21'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/20'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/19'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/18'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/17'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/16'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/15'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/14'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/13'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/12'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/11'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/10'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/9'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/8'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/7'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/6'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/5'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/4'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/32'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/31'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/28'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/27'
2015-09-01 19:49:59: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/26
syslog:
Sep  1 20:05:20 meye kernel: [  994.527425] uvcvideo: Failed to set UVC probe control : -110 (exp. 26).
Sep  1 20:05:20 meye motion: [1] Retrying until successful connection with camera
Sep  1 20:05:20 meye motion: [1] cap.driver: "uvcvideo"
Sep  1 20:05:20 meye motion: [1] cap.card: "UVC Camera (046d:08cf)"
Sep  1 20:05:20 meye motion: [1] cap.bus_info: "usb-20980000.usb-1.3.1"
Sep  1 20:05:20 meye motion: [1] cap.capabilities=0x84200001
Sep  1 20:05:20 meye motion: [1] - VIDEO_CAPTURE
Sep  1 20:05:20 meye motion: [1] - STREAMING
Sep  1 20:05:25 meye motion: [1] Config palette index 8 (YU12) doesn't work.

also the raspicam sometimes loses connection.

Any suggestion?
Thanks

ccrisan, 2015-09-02:
Have you installed motionEye 0.27 from scratch (and also removed the old config files)?

sgiannulli, 2015-09-02:
just finished a fresh install from scratch and:
usb webcam still not working with the same errors as above
raspicam produces the following output in motioneye.log:
2015-09-02 16:05:28: [motioneye]     INFO: hello! this is motionEye server 0.27.1
2015-09-02 16:05:32: [motioneye]     INFO: cleanup started
2015-09-02 16:05:32: [motioneye]     INFO: wsswitch started
2015-09-02 16:05:32: [motioneye]     INFO: thumbnailer started
2015-09-02 16:05:32: [motioneye]     INFO: server started
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/32'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/31'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/28'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/27'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/26'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/25'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/24'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/23'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/22'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/21'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/20'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/19'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/18'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/17'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/16'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/15'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/14'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/13'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/12'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/11'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/10'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/9'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/8'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/7'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/6'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/5'
2015-09-02 16:06:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/4'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/32'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/31'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/28'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/27'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/26'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/25'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/24'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/23'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/22'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/21'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/20'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/19'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/18'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/17'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/16'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/15'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/14'
[... cut ...]
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/8'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/7'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/6'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/5'
2015-09-02 16:07:02: [motioneye]    ERROR: stat failed: [Errno 2] No such file or directory: '/run/udev/watch/4'

ccrisan, 2015-09-02:
What have you set for *File Storage/Root Directory*?

sgiannulli, 2015-09-02:
yes......no more errors with raspicam as soon as different storage path set
usb webcam: I'm now wondering it could be something related to the new kernel on Raspbian recently released..........you can close this issue
Thanks

closed 2015-09-02 by sgiannulli