motioneye-project/motioneye #5: Running motioneye as non-root

tschabha, 2015-09-02:
With previous versions is was easy to start the service as non-root user. How can it be done since ver. 0.27 where meyectl is used for controlling the service?

ccrisan, 2015-09-02:
I planned to write a small note on this somewhere in the install instructions. Basically you'll need to run `meyectl startserver -c /home/user/motioneye/conf/motioneye.conf` with your user. The example assumes that `/home/user/motioneye/` is writable by your user and that the given `motioneye.conf` has paths configured to writable directories as well. (I recommend using `/home/user/motioneye/conf|log|run|media`).

tschabha, 2015-09-02:
Thanx! It works - almost - fine!
I've got some error messages in the log (see the log below). And some icons are missing from the UI for "settings", "switch user", "motioneye home". Apart from this it looks all right.
I'm running it on Arch Linux - CuBox-i4 Pro...

------------------- motion.log:
[0] Processing thread 0 - config file /home/motioneye/motion.conf
[0] Processing config file thread-1.conf
[0] Unknown config option "locate_motion_style"
[0] Motion 3.2.12 Started
[0] ffmpeg LIBAVCODEC_BUILD 3489124 LIBAVFORMAT_BUILD 3481700
[0] Thread 1 is from thread-1.conf
[1] Thread 1 started
[1] cap.driver: "uvcvideo"
[0] motion-httpd/3.2.12 running, accepting connections
[1] cap.card: "UVC Camera (046d:0804)"
[0] motion-httpd: waiting for data on port TCP 7999
[1] cap.bus_info: "usb-ci_hdrc.0-1.1"
[1] cap.capabilities=0x84000001
[1] - VIDEO_CAPTURE
[1] - STREAMING
[1] Config palette index 8 (YU12) doesn't work.
[1] Supported palettes:
[1] 0: YUYV (YUV 4:2:2 (YUYV))
[1] 1: MJPG (MJPEG)
[1] Selected palette YUYV
[1] Test palette YUYV (640x480)
[1] Using palette YUYV (640x480) bytesperlines 1280 sizeimage 614400 colorspace 00000008
[1] found control 0x00980900, "Brightness", range 0,255
[1]     "Brightness", default 128, current 221
[1] found control 0x00980901, "Contrast", range 0,255
[1]     "Contrast", default 32, current 32
[1] found control 0x00980902, "Saturation", range 0,255
[1]     "Saturation", default 32, current 32
[1] found control 0x00980913, "Gain", range 0,255
[1]     "Gain", default 0, current 253
[1] mmap information:
[1] frames=4
[1] 0 length=614400
[1] 1 length=614400
[1] 2 length=614400
[1] 3 length=614400
[1] Using V4L2
[1] Resizing pre_capture buffer to 1 items
[1] Started stream webcam server in port 8081
[1] Resizing pre_capture buffer to 3 items
Traceback (most recent call last):
  File "/usr/bin/meyectl", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3084, in <module>
    @_call_aside
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3070, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3097, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 651, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 952, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 847, in resolve
    new_requirements = dist.requires(req.extras)[::-1]
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2602, in requires
    dm = self._dep_map
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2803, in _dep_map
    self.__dep_map = self._compute_dependencies()
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2820, in _compute_dependencies
    from _markerlib import compile as compile_marker
  File "/usr/lib/python2.7/site-packages/_markerlib/__init__.py", line 3, in <module>
    from _markerlib.markers import default_environment, compile, interpret
  File "/usr/lib/python2.7/site-packages/_markerlib/markers.py", line 46, in <module>
    'platform.version': platform.version(),
[... 78 lines cut ...]
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 3097, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 651, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 952, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 847, in resolve
    new_requirements = dist.requires(req.extras)[::-1]
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2602, in requires
    dm = self._dep_map
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2803, in _dep_map
    self.__dep_map = self._compute_dependencies()
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2820, in _compute_dependencies
    from _markerlib import compile as compile_marker
  File "/usr/lib/python2.7/site-packages/_markerlib/__init__.py", line 3, in <module>
    from _markerlib.markers import default_environment, compile, interpret
  File "/usr/lib/python2.7/site-packages/_markerlib/markers.py", line 46, in <module>
    'platform.version': platform.version(),
  File "/usr/lib/python2.7/platform.py", line 1338, in version
    return uname()[3]
  File "/usr/lib/python2.7/platform.py", line 1277, in uname
    processor = _syscmd_uname('-p','')
  File "/usr/lib/python2.7/platform.py", line 1012, in _syscmd_uname
    rc = f.close()
IOError: [Errno 10] No child processes

ccrisan, 2015-09-02:
Weird. Have you stopped the root process completely? Have you reinstalled the new user-based motionEye from scratch? Are you sure all the permissions are correct when it comes to your `/home/user/motionEye` folder?

closed 2015-12-26 by ccrisan