motioneye-project/motioneye #6: ERROR: decoder jpeg not available

sgiannulli, 2015-09-03:
Hi Calin,
is this error on motioneye log (opening interval snapshot images) telling some package is missing? I'm on Raspbian
Thanks
2015-09-03 18:39:18: [motioneye]    ERROR: 500 GET /picture/1/preview/2015-09-03/18-30-00.jpg?height=51&_username=admin&_signature=f6e98f4ddee3c605b73786eaa1d3e49556299ded (192.168.0.21) 106.85ms
2015-09-03 18:39:19: [motioneye]    ERROR: decoder jpeg not available
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1369, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1572, in wrapper
    result = method(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 716, in get
    self.preview(camera_id, filename)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 121, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 875, in preview
    height=self.get_argument('height', None))
  File "/usr/local/lib/python2.7/dist-packages/motioneye/mediafiles.py", line 676, in get_media_preview
    image.thumbnail((width, height), Image.LINEAR)
  File "/usr/local/lib/python2.7/dist-packages/PIL/Image.py", line 1814, in thumbnail
    im = self.resize(size, resample)
  File "/usr/local/lib/python2.7/dist-packages/PIL/Image.py", line 1557, in resize
    self.load()
  File "/usr/local/lib/python2.7/dist-packages/PIL/ImageFile.py", line 203, in load
    d = Image._getdecoder(self.mode, d, a, self.decoderconfig)
  File "/usr/local/lib/python2.7/dist-packages/PIL/Image.py", line 420, in _getdecoder
    raise IOError("decoder %s not available" % decoder_name)
IOError: decoder jpeg not available

ccrisan, 2015-09-04:
Please `apt-get install libjpeg-dev`, then `pip uninstall -y pillow` and then `pip install pillow`. Let me know if this solves your problem. I'll update the wiki instructions accordingly.

sgiannulli, 2015-09-04:
yes, the error has gone away.
Thanks

closed 2015-09-04 by ccrisan