motioneye-project/motioneye #7: meyectl relayevent: error: unrecognized arguments: -b

sgiannulli, 2015-09-04:
Hello Calin,
sorry to bother once again; for some reason I have "Motion Triggered" not working in "Still Images" and motion.log reports the following error:
meyectl relayevent: error: unrecognized arguments: -b
usage: meyectl relayevent [-c CONFIG_FILE] [-d] [-h] [-l] [-v] event thread_id
meyectl relayevent: error: unrecognized arguments: -b
don't know if relevant for motion triggered; I'm using motion 3.2.12.
thanks

ccrisan, 2015-09-04:
Yes, it is relevant and you found a special case that doesn't work. Thanks for reporting it, I'll make sure to fix this asap.

sgiannulli, 2015-09-04:
Thank you; will it affect also some other option? I mean standard motion detection, email sending...?
I will wait for your fix to go on
Thanks again

ccrisan, 2015-09-04:
The bug should be fixed with the latest commit. I'll release 0.27.2  later tonight.

closed 2015-09-04 by ccrisan