motioneye-project/motioneye #8: Fast Network Camera support?

IndrekHaav, 2015-09-05:
Is motioneye supposed to support Fast Network Camera (using streameye), or is that only a MotionPie thing?

I'm running motioneye 0.27.2 on a RPi 2 with Raspbian, with the NoIR camera module, and cannot see the option to enable Fast Network Camera anywhere in the settings.

ccrisan, 2015-09-07:
Fast Network Camera is a motionPie-specific thing. It's `motionEye` + `streamEye` + some extra config stuff, instead of `motionEye` + `motion`.

You can however "reproduce" the Fast Network Camera yourself on a Raspberry PI runnning Raspbian.  Just read [streamEye's readme](https://github.com/ccrisan/streameye#streameye) and you'll figure it out how.

closed 2015-09-07 by ccrisan

IndrekHaav, 2015-09-07:
Thanks, I'll look into that.

Not having the UI is no biggie, I have motionEye running on a separate virtual server that acts like a hub, so the RPi doesn't actually need to do anything besides capturing JPEGs and serving them as a MJPG stream.

timdonovanuk, 2017-07-05:
Damn, this answer yet again...wondering why so much differs between motionPie and motionEye. MotionPie should just be a ready to rock .img.