motioneye-project/motioneye #9: How to Start/Stop motioneye?

Samfox2, 2015-09-06:
I combined the "old" motioneye versions as a daemon and combined them with my doorbell. When smb. rang the door motioneye daemon was stopped, a picture was taken via command line tool and send per telegram to my mobile and after that motioneye daemon was started again. Worked really well. Now with all the changes this doesn't work anymore as paths and the tool itself have changed. How can I stop/start the motioneye (meyectl?) with command line?

ccrisan, 2015-09-07:
Of course there is a way to start and stop it from command line. I should write a small wiki entry on this topic, but until then, `meyectl -h` should show a basic usage. Depending on how you configured your motionEye, the commands should look something like:

    meyectl startserver -b -c /path/to/motioneye.conf

and

    meyectl stopserver -b -c /path/to/motioneye.conf

The `-b` stands for running in background. If you omit it, it will start in foreground and it's your responsibility to kill it (the `stopserver` command no longer makes sense).

closed 2015-09-07 by ccrisan

Samfox2, 2015-09-07:
That should do the job, thx!

Morphy99, 2019-08-11:
I'm trying to attempt something similar but with motioneye set to Fast Network Camera. I want to stop it from streaming and turn it on remotely. I've been doing this with a smart plug but I don't want to corrupt the SD card so was going to set a script to SSH and shutdown then switch the plug off however I can't shutdown meye from ssh?

I tried the above but if I do `meyectl stopserver -b -c /path/to/motioneye.conf` the stream continues however I cant reach the GUI anymore. I had a look at the other issue referenced however I can't find any  thread-*.conf files located in /data/etc/. I'm assuming as this is regarding disabling them on a meye server pi.

CRImier, 2019-08-12:
@Morphy99 why don't you just do a `poweroff`? That should shut down the entire system and actually write all the data that hasn't been written to the SD card yet, simply stopping the MotionEye server won't help much. Otherwise, my experience is that you can use `sudo systemctl stop motioneye` to stop *everything* if you're launching motioneye as a service - doesn't seem to be what you actually need, but figured I'd post this just in case.

Morphy99, 2019-08-13:
> why don't you just do a `poweroff`? 
Ahh thanks I wasn't aware of that command I was trying 'shutdown -h now' which doesnt work on meyeos. 

> Otherwise, my experience is that you can use `sudo systemctl stop motioneye` to stop _everything_ if you're launching motioneye as a service

Yea this doesn't work on meyeos either as systemctl isn't compiled. I'd like to be able to stop the stream without powering off the rpi as I wouldn't need a smart plug then and it would also be faster.

CRImier, 2019-08-13:
Ah, you're using meyeos, got it. I don't actually have SSH enabled on any of my meyeos-powerd Pi, so I can't really check any other options. I figure the actual stream is being done by the `motion` executable, any chance the `meyectl` command can start/stop that, too? I can't check ATM, but you might figure something out.

Morphy99, 2019-08-13:
I think the stream is being done by streameye as thats whats consuming the CPU in top. I can't find the correct .conf in /etc/data to use meyectl as just stopserver stops the GUI and not the stream component. 

So running the poweroff command has been with its problems, I get no such file or directory when executing via ssh command. I added the /sbin/poweroff path and this works but not from a home assistant shell command as this is how I'm looking to execute it. I think it's because the SSH connection doesn't close gracefully. If I run this script from the terminal it exits ok but I get "return code: 127" in home assistant:
`#!/bin/sh

ssh root@192.168.xxx.xxx "/sbin/poweroff &&exit"`