motioneye-project/motioneye #11: pip command to install latest commit

sgiannulli, 2015-09-08:
Hi Calin,
0.27.2 version still giving me lot of errors; is there a pip command to install the latest commit before a new version is released?
Thanks

ccrisan, 2015-09-08:
The difference between 0.27.2 is insignificant. First of all, **let me see what errors you are talking about** so I can identify the source of the problem.

Secondly, to answer your question, there is a possibility to install the latest commit:

1. Download the latest snapshot: https://github.com/ccrisan/motioneye/archive/master.tar.gz.
2. Uninstall the existing version: `pip uninstall motioneye`.
3. Unpack the archive: `tar zxvf master.tar.gz`.
4. Install motionEye: `cd [extracted_motioneye_folder] && python setup.py install --prefix=/usr`.

sgiannulli, 2015-09-08:
motioneye is not recording any movie when motion is detected; I have compiled Mr Dave's fork and I got the following warnings in motion.log:
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_output_movies"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_variable_bitrate"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_video_codec"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_filename"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_bps"
[0] [NTC] [ALL] config_thread: Processing config file thread-2.conf
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_output_movies"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_variable_bitrate"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_video_codec"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_filename"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_bps"
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-d71cebf Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (EMG)
now I don't know if it's a motioneye problem about how it reads motion.conf parameters ( found it here http://www.lavrsen.dk/foswiki/bin/view/Motion/SupportQuestion2014x05x06x220454 ).
Then ( and this is quite trivial ) when an email is received after motion detection event, outlook 2007 reports as email recipient " n; a; m; e;.s; u r; n; a; m; e; @; g; m; a; i; l; .; c; o; m "........why all those semicolons? 
Thanks

ccrisan, 2015-09-08:
I think you have upgraded from motionEye 0.26 or earlier. You *must* remove all your config files and re-add the cameras after installing motionEye 0.27.

closed 2015-09-08 by ccrisan

sgiannulli, 2015-09-08:
yes, seemed I forgot to completely purge remove old motion install before compiling from Mr Dave fork.
If you're interested on, you could put a link on motion compiling page for Raspbian users:
https://crcok.wordpress.com/2015/01/22/cant-compile-motion-with-rtsp-on-raspberry-pi/
doing the following before compiling is the way to avoid ffmpeg support to be disabled:
sudo ln -s /usr/lib/arm-linux-gnueabihf/libavcodec.a /usr/lib/libavcodec.a
sudo ln -s /usr/lib/arm-linux-gnueabihf/libavcodec.so /usr/lib/libavcodec.so
sudo ln -s /usr/lib/arm-linux-gnueabihf/libavformat.a /usr/lib/libavformat.a
sudo ln -s /usr/lib/arm-linux-gnueabihf/libavformat.so /usr/lib/libavformat.so
Thanks

ccrisan, 2015-09-09:
The instructions are [here](https://github.com/ccrisan/motioneye/wiki/Compiling-Motion) and are applicable to various common distros (including Raspbian). Symlinking doesn't seem to be required.

sgiannulli, 2015-09-09:
When I compiled Mr Dave motion the first time I exactly faced the error reported in the link I gave ( on the latest version of Raspbian ); creating all above symlinks did the trick for me. Anyway it is just a suggestion for users may face the same problem.
Thanks

electronicamystica, 2015-09-17:
@sgiannulli : thanks!

I also had to link the files to get ffmpeg support using rapsbian on vB

ccrisan, 2015-09-18:
I have updated both [Install On Raspbian](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) and [Compiling Motion](https://github.com/ccrisan/motioneye/wiki/Compiling-Motion) to help users build motion on Raspbian.

More precisely, instead of creating links to the misplaced libs, the configure script needs to be told to look for them in some other places:

    ./configure --prefix=/usr --with-ffmpeg=/usr/lib/arm-linux-gnueabihf  --with-ffmpeg-headers=/usr --without-pgsql --without-sdl --without-sqlite3 --without-mysql