motioneye-project/motioneye #12: delete of /var/lib/motioneye

sgiannulli, 2015-09-08:
Hi,
strange behaviour for me is that if you delete pictures or movies from webUI motioneye also deletes the folder /var/lib/motioneye; then if a picture or a movie is taken it creates the folder back again
can you replicate it?
Thanks

ccrisan, 2015-09-09:
What's your root directory?

sgiannulli, 2015-09-09:
I was testing snapshots every ten minutes with one camera and I left /var/lib/motioneye; it creates /var/lib/motioneye/date-of-today/jpg1....jpg2.......When you choose "delete all pictures taken" also /var/lib/motioneye is deleted. Not yet tried with a complete different path, but maybe this is what it has to do.
Thanks

ccrisan, 2015-09-09:
Yeap, that's the expected (not necessarily desired) behaviour. I will consider preventing removing the root directory though.

electronicamystica, 2015-09-18:
I stumbled on the same issue resulting in motioneye refusing to start since the directory was missing.
The log files gave me some clarity here.

ccrisan, 2015-09-19:
Fixed with the latest commit. motionEye will create a `.keep` sentinel file in the root directory to make sure it never gets removed.

closed 2015-09-19 by ccrisan