motioneye-project/motioneye #13: Message: An error occurred. Refreshing recommended.

kdeblois, 2015-09-14:
Using 0.27.2 on a raspberrypi2...

I started getting this message right after upgrading motion from what is in the wheezy distribution to the Mr-Dave fork version. I am no longer able to add and save any camera and keep getting this message. I did uninstall the old version before compiling the new.

No camera configuration works. I had used http://192.168.2.9:88/videoSub with user name and password in the appropriate boxes.

By the way, I did have the camera working in 0.27.2 (but not with rtsp) with the old version of motion from the distribution.

As this tool does not seem to allow me to attach a text file, let me copy and paste part of the log file:

==========================================================
2015-09-13 23:06:17: [motioneye]    ERROR: 500 GET /config/list/?_=1442178377588 (127.0.0.1) 12.35ms
2015-09-13 23:10:43: [motioneye]    ERROR: 'stream_localhost'
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1369, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1572, in wrapper
    result = method(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 175, in get
    self.list()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 121, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 590, in list
    ui_config = config.motion_camera_dict_to_ui(local_config)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 929, in motion_camera_dict_to_ui
    'video_streaming': not data['stream_localhost'],
KeyError: 'stream_localhost'
2015-09-13 23:10:43: [motioneye]    ERROR: 500 GET /config/list/?_=1442178643356 (127.0.0.1) 12.14ms
2015-09-13 23:25:35: [motioneye]    ERROR: 'stream_localhost'
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1369, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1572, in wrapper
    result = method(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 175, in get
    self.list()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 121, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 590, in list
    ui_config = config.motion_camera_dict_to_ui(local_config)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 929, in motion_camera_dict_to_ui
    'video_streaming': not data['stream_localhost'],
KeyError: 'stream_localhost'
2015-09-13 23:25:35: [motioneye]    ERROR: 500 GET /config/list/?_=1442179535568&_username=admin&_signature=8628f341bfbc29406fe41b04ce2563d0e3f407f6 (127.0.0.1) 12.75ms
2015-09-13 23:27:03: [motioneye]     INFO: adding new camera with id 8...
2015-09-13 23:27:10: [motioneye]    ERROR: 'stream_localhost'
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1369, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 1572, in wrapper
    result = method(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 175, in get
    self.list()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 121, in wrapper
    return func(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/handlers.py", line 590, in list
    ui_config = config.motion_camera_dict_to_ui(local_config)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 929, in motion_camera_dict_to_ui
    'video_streaming': not data['stream_localhost'],
KeyError: 'stream_localhost'
2015-09-13 23:27:10: [motioneye]    ERROR: 500 GET /config/list/?_=1442179629052&_username=admin&_signature=5715499591c41e3d0e41483e828611078ffaee4f (127.0.0.1) 11.67ms
2015-09-13 23:33:24: [motioneye]     INFO: interrupt signal received, shutting down...
2015-09-13 23:33:24: [motioneye]     INFO: server stopped
2015-09-13 23:33:27: [motioneye]     INFO: motion stopped
2015-09-13 23:33:27: [motioneye]     INFO: bye!
2015-09-13 23:33:27: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
2015-09-13 23:33:27: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
2015-09-13 23:33:27: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
2015-09-13 23:33:27: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
2015-09-13 23:33:27: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
2015-09-13 23:33:27: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
2015-09-13 23:33:27: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 111] Connection refused
2015-09-13 23:33:27: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 111] Connection refused
2015-09-13 23:33:45: [motioneye]     INFO: hello! this is motionEye server 0.27.2
2015-09-13 23:33:45: [motioneye]     INFO: motion not installed

kdeblois, 2015-09-14:
Sorry, I had used: rtsp://192.168.2.9:88/videoSub + credentials in appropriate boxes.

The error message disappeared after I deleted the existing thread-X.conf files from /etc/motioneye, so one problem solved!

I am now getting 'unable to open video device' + grey camera screen. So, rtsp is not working yet. I do get images when I configure the camera as http (jpeg) camera using the string:

http://192.168.2.9:88/cgi-bin/CGIProxy.fcgi?cmd=snapPicture2&usr=XXXX&pwd=XXXX.

Any advice re. rtsp configuration?

ccrisan, 2015-09-14:
1) Have you installed 0.27 from scratch (including the configuration) or have you somehow upgraded from a previous version? If you used the `motion.conf` and `thread-x.conf` files from an older version, then these problems are expected.

2) If you started everything from scratch, please try to reproduce the problem again and write down a few steps for me to follow in order to be able to reproduce the problem myself.

3) If you run "motion -h" in a terminal, what version does it report?

kdeblois, 2015-09-14:
Thanks for your help/

The version of motion I am using is Unofficial-Git-d71cebf

I indeed compiled motion from scratch after uninstalling the wheezy distribution version

No more error messages. I start-up motioneye server from the command line by running: /etc//init.d//motioneye start

In my browser I go to localhost:8765, the program opens and I log in as admin.

Two camera screens appear pointing to one and the same camera, the FI9853EP. One is configured as rtsp and the second to http with the strings I gave you earlier in the thread.

The first camera screen is grey with an 'unable to open video device' message, the second gives me a good picture.

ccrisan, 2015-09-14:
The fact that `2015-09-13 23:33:45: [motioneye] INFO: motion not installed` appeared in your log made me think that motion could not be available for some reason.

Please send me the `motion.log` file, so that I can find out why your RTSP camera doesn't work.

kdeblois, 2015-09-14:
That motioneye message no longer appears in the motioneye log file.

What I am getting in the motion.log is the following:

[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-2.conf
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_output_movies"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_variable_bitrate"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_video_codec"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_bps"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_filename"
[0] [ALR] [ALL] config_thread: Thread config file thread-3.conf not found: No such file or directory
[0] [ALR] [ALL] config_thread: Thread config file thread-4.conf not found: No such file or directory
[0] [ALR] [ALL] config_thread: Thread config file thread-5.conf not found: No such file or directory
[0] [ALR] [ALL] config_thread: Thread config file thread-6.conf not found: No such file or directory
[0] [ALR] [ALL] config_thread: Thread config file thread-7.conf not found: No such file or directory
[0] [ALR] [ALL] config_thread: Thread config file thread-8.conf not found: No such file or directory
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_output_movies"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_variable_bitrate"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_video_codec"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_bps"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_filename"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "control_port"
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-d71cebf Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (EMG)

ccrisan, 2015-09-14:
After installing the new motion, have you restarted motionEye and re-added your cameras? It looks as though motionEye is still using the old directives in the config files.

kdeblois, 2015-09-14:
I actually did. I compiled motion using your wiki instructions, and even
did a pip uninstall motioneye, followed by a fresh install. This is the
latest from the motioneye.log file. It does report an relay event error...

Should I make changes in my config files?

==============================================================================

2015-09-14 12:52:28: [motioneye]     INFO: interrupt signal received,
shutting down...
2015-09-14 12:52:28: [motioneye]     INFO: server stopped
2015-09-14 12:52:30: [motioneye]     INFO: motion stopped
2015-09-14 12:52:30: [motioneye]     INFO: bye!
2015-09-14 12:52:30: [relayevent]    ERROR: failed to relay event: [Errno
socket error] [Errno 104] Connection reset by peer
2015-09-14 12:52:30: [relayevent]    ERROR: failed to relay event: [Errno
socket error] [Errno 104] Connection reset by peer
2015-09-14 12:52:31: [motioneye]     INFO: hello! this is motionEye server
0.27.2
2015-09-14 12:52:33: [motioneye]     INFO: cleanup started
2015-09-14 12:52:33: [motioneye]     INFO: wsswitch started
2015-09-14 12:52:33: [motioneye]     INFO: thumbnailer started
2015-09-14 12:52:33: [motioneye]     INFO: server started
2015-09-14 13:20:36: [motioneye]     INFO: interrupt signal received,
shutting down...
2015-09-14 13:20:36: [motioneye]     INFO: server stopped
2015-09-14 13:20:38: [motioneye]     INFO: motion stopped
2015-09-14 13:20:38: [motioneye]     INFO: bye!
2015-09-14 13:20:38: [relayevent]    ERROR: failed to relay event: [Errno
socket error] [Errno 104] Connection reset by peer
2015-09-14 13:20:38: [relayevent]    ERROR: failed to relay event: [Errno
socket error] [Errno 104] Connection reset by peer
2015-09-14 13:20:39: [motioneye]     INFO: hello! this is motionEye server
0.27.2
2015-09-14 13:20:41: [motioneye]     INFO: cleanup started
2015-09-14 13:20:41: [motioneye]     INFO: wsswitch started
2015-09-14 13:20:41: [motioneye]     INFO: thumbnailer started
2015-09-14 13:20:41: [motioneye]     INFO: server started

-- 

This email and any files sent with it are confidential and intended only 
for the person it is addressed to. If you have received this message in 
error please let us know immediately and delete it. The content of this 
message belongs to the sender and the views expressed do not necessarily 
reflect the views of United Bible Societies. 

United Bible Societies Association, a company limited by guarantee. 
Registered in England and Wales No 2264875. Registered Charity No 800058 
Registered Office: Stonehill Green, Westlea, Swindon, Wiltshire, SN5 7PJ, 
England.

ccrisan, 2015-09-14:
There's a reference to a thread-8.conf (!) in your logs. Please remove all the thread-x.conf files as well as motion.conf and re-add everything from scratch. If the problem still persists, I'd need some steps to reproduce it.

kdeblois, 2015-09-14:
I have followed your instructions, deleted the thread-x.config files and the motion.conf. I added the rtsp camera with the rtsp settings and still no image. 
This is motion.log.....

[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_output_movies"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_variable_bitrate"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_video_codec"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_bps"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_filename"
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-d71cebf Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (EMG)

and this is motioneye.log:

2015-09-14 14:57:30: [motioneye]     INFO: interrupt signal received, shutting down...
2015-09-14 14:57:30: [motioneye]     INFO: server stopped
2015-09-14 14:57:33: [motioneye]     INFO: motion stopped
2015-09-14 14:57:33: [motioneye]     INFO: bye!
2015-09-14 14:57:33: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
2015-09-14 14:57:34: [motioneye]     INFO: hello! this is motionEye server 0.27.2
2015-09-14 14:57:36: [motioneye]     INFO: cleanup started
2015-09-14 14:57:36: [motioneye]     INFO: wsswitch started
2015-09-14 14:57:36: [motioneye]     INFO: thumbnailer started
2015-09-14 14:57:36: [motioneye]     INFO: server started

Kindly advise what I can do next. Thanks for your patience! I have rebooted my camera, just in case....

[image]

ccrisan, 2015-09-14:
1) Do you have `motion_binary` set in your `motioneye.conf`?
2) If you run `motion -h` in your terminal what is the output?
3) Please enable debugging and repeat the operation:

* stop your motionEye server
* set `log_level` to `debug` in your `motioneye.conf` file
* start your motionEye server
* re-add the cameras
* send me the new log `motioneye.log` file (this time it should contain debugging details)

kdeblois, 2015-09-14:
OK.

Re 1) It was commented out (automatic detection), but I enabled it with the right path (/usr/local/bin/motion)

2) It still says: Version Unofficial-Git-d71cebf, I did not make any changes.

3) Here is the result:
2015-09-14 15:46:17: [motioneye]     INFO: hello! this is motionEye server 0.27.2
2015-09-14 15:46:17: [motioneye]    DEBUG: reading main config from file /etc/motioneye/motion.conf...
2015-09-14 15:46:17: [motioneye]    DEBUG: loading additional config structure for main, without separators
2015-09-14 15:46:17: [motioneye]    DEBUG: additional config section: network
2015-09-14 15:46:17: [motioneye]    DEBUG: listing config dir /etc/motioneye...
2015-09-14 15:46:17: [motioneye]    DEBUG: found camera with id 1
2015-09-14 15:46:17: [motioneye]    DEBUG: reading camera config from /etc/motioneye/thread-1.conf...
2015-09-14 15:46:17: [motioneye]    DEBUG: loading additional config structure for camera, without separators
2015-09-14 15:46:17: [motioneye]    DEBUG: starting motion
2015-09-14 15:46:17: [motioneye]    DEBUG: using motion binary "/usr/local/bin/motion"
2015-09-14 15:46:19: [motioneye]     INFO: cleanup started
2015-09-14 15:46:19: [motioneye]     INFO: wsswitch started
2015-09-14 15:46:19: [motioneye]     INFO: thumbnailer started
2015-09-14 15:46:19: [motioneye]     INFO: server started
2015-09-14 15:46:25: [motioneye]    DEBUG: loading additional config structure for main, with separators
2015-09-14 15:46:25: [motioneye]    DEBUG: additional config section: network
2015-09-14 15:46:25: [motioneye]    DEBUG: loading additional config structure for camera, with separators
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET / (192.168.2.254) 1154.59ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/css/jquery.timepicker.css (192.168.2.254) 25.04ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/css/ui.css (192.168.2.254) 3.56ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/css/main.css (192.168.2.254) 3.78ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/js/css-browser-selector.js (192.168.2.254) 3.57ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/js/jquery.min.js (192.168.2.254) 4.97ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/js/jquery.timepicker.min.js (192.168.2.254) 3.54ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/js/jquery.mousewheel.js (192.168.2.254) 3.70ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/js/ui.js (192.168.2.254) 3.87ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/js/main.js (192.168.2.254) 5.78ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/img/motioneye-logo.svg (192.168.2.254) 3.38ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/fnt/mavenpro-bold-webfont.woff (192.168.2.254) 3.80ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/img/logout.svg (192.168.2.254) 3.49ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/fnt/mavenpro-regular-webfont.woff (192.168.2.254) 3.80ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/img/apply-progress.gif (192.168.2.254) 3.93ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 200 GET /login/?_=1442238384685&_username=admin&_signature=ce85e1b5780bc4e014038a7dd0fd50e60d9a89d5 (192.168.2.254) 3.94ms
2015-09-14 15:46:26: [motioneye]    DEBUG: getting main config
2015-09-14 15:46:26: [motioneye]    DEBUG: 200 GET /config/main/get/?_=1442238384707&_username=admin&_signature=bd901fe97d11e85661092be6f6f13101896ff653 (192.168.2.254) 4.21ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 200 GET /static/favicon.ico (192.168.2.254) 23.69ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/img/settings.svg (192.168.2.254) 3.96ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/img/main-loading-progress.gif (192.168.2.254) 3.63ms
2015-09-14 15:46:26: [motioneye]    DEBUG: 304 GET /static/img/combo-box-arrow.svg (192.168.2.254) 3.71ms
2015-09-14 15:46:26: [motioneye]    DEBUG: listing cameras
2015-09-14 15:46:26: [motioneye]    DEBUG: listing disks using /dev/disk/by-id/
2015-09-14 15:46:26: [motioneye]    DEBUG: found partition "2" at "/dev/mmcblk0p2" on bus "memstick": "SE08G 0xc6e15aa1"
2015-09-14 15:46:26: [motioneye]    DEBUG: found partition "1" at "/dev/mmcblk0p1" on bus "memstick": "SE08G 0xc6e15aa1"
2015-09-14 15:46:26: [motioneye]    DEBUG: found disk at "/dev/mmcblk0" on bus "memstick": "SE08G 0xc6e15aa1"
2015-09-14 15:46:26: [motioneye]    DEBUG: listing mounts...
2015-09-14 15:46:26: [motioneye]    DEBUG: found mount "/dev/root" at "/"
2015-09-14 15:46:26: [motioneye]    DEBUG: found mount "devtmpfs" at "/dev"
2015-09-14 15:46:26: [motioneye]    DEBUG: found mount "tmpfs" at "/run"
2015-09-14 15:46:26: [motioneye]    DEBUG: found mount "proc" at "/proc"
2015-09-14 15:46:26: [motioneye]    DEBUG: found mount "sysfs" at "/sys"
2015-09-14 15:46:26: [motioneye]    DEBUG: found mount "devpts" at "/dev/pts"
2015-09-14 15:46:26: [motioneye]    DEBUG: found mount "/dev/mmcblk0p1" at "/boot"
2015-09-14 15:46:26: [motioneye]    DEBUG: found mount "gvfs-fuse-daemon" at "/root/.gvfs"
2015-09-14 15:46:26: [motioneye]    DEBUG: found mount "fusectl" at "/sys/fs/fuse/connections"
2015-09-14 15:46:26: [motioneye]    DEBUG: getting disk usage for path /var/lib/motioneye...
2015-09-14 15:46:26: [motioneye]    DEBUG: 200 GET /config/list/?_=1442238384878&_username=admin&_signature=853460ee56de229baa6f2c8deae1dfda11945d7e (192.168.2.254) 18.63ms
[... cut ...]
2015-09-14 15:47:11: [relayevent]    DEBUG: hello!
2015-09-14 15:47:11: [relayevent]    DEBUG: event = stop
2015-09-14 15:47:11: [relayevent]    DEBUG: thread_id = 1
2015-09-14 15:47:11: [relayevent]    DEBUG: reading main config from file /etc/motioneye/motion.conf...
2015-09-14 15:47:12: [motioneye]     INFO: motion stopped
2015-09-14 15:47:12: [motioneye]     INFO: bye!
2015-09-14 15:47:12: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
2015-09-14 15:47:12: [relayevent]    DEBUG: bye!

ccrisan, 2015-09-14:
Could you set up SSH remote access for me on your server so I can try to reproduce the problem?

kdeblois, 2015-09-14:
Sure, but I do not know if I can do that without a bit of help from your side.

kdeblois, 2015-09-14:
I found some instructions and managed to install SSH server. Now I guess I have to forward port 22 to the raspberrypi2, correct?

remkolodder, 2015-09-14:
Yes that is correct. Pls share the information privately with christian ...

Cheers,

Remko Lodder
 /* sent from my phone and thus brief and to the point *\

kdeblois, 2015-09-14:
> <javascript:_e(%7B%7D,'cvml','notifications@github.com');>> het volgende

Thanks, I just did.

-- 

This email and any files sent with it are confidential and intended only 
for the person it is addressed to. If you have received this message in 
error please let us know immediately and delete it. The content of this 
message belongs to the sender and the views expressed do not necessarily 
reflect the views of United Bible Societies. 

United Bible Societies Association, a company limited by guarantee. 
Registered in England and Wales No 2264875. Registered Charity No 800058 
Registered Office: Stonehill Green, Westlea, Swindon, Wiltshire, SN5 7PJ, 
England.

ccrisan, 2015-09-18:
@kdeblois the problem with your setup was the fact that *motion* was compiled without *ffmpeg* support. This was due to the fact that on Raspbian, ffmpeg files are installed in non-standard locations (as far as I can tell).

As a consequence:
1. I updated the [Install On Raspbian](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) page to help users properly configure their motion build.
2. I updated the [Compiling Motion](https://github.com/ccrisan/motioneye/wiki/Compiling-Motion) page with a small note regarding Raspbian.
3. I have prepared a prebuilt motion binary (Mr Dave's fork, of course) for Raspbian; you can download it from [here](https://github.com/ccrisan/motioneye/wiki/motion-binaries/motion-mrdave-raspbian).

closed 2015-09-18 by ccrisan