motioneye-project/motioneye #14: motioneye not starting

karma007, 2015-09-15:
hello, i'm on rasbian jessie on RPI2.
First command does not work : 
pi@karmapi ~ $ sudo /etc/init.d/motioneye start
[....] Starting motioneye (via systemctl): motioneye.serviceJob for motioneye.service failed. See 'systemctl status motioneye.service' and 'journalctl -xn' for details.
 failed!

trying this one give me this error : 
pi@karmapi ~ $ sudo systemctl status motioneye.service
● motioneye.service - LSB: Start the motionEye server
   Loaded: loaded (/etc/init.d/motioneye)
   Active: failed (Result: exit-code) since mar. 2015-09-15 14:20:31 CEST; 33s ago
  Process: 4452 ExecStart=/etc/init.d/motioneye start (code=exited, status=1/FAILURE)

sept. 15 14:20:31 karmapi motioneye[4452]: Starting motionEye server:CRITICAL:root:failed to read settings from "/etc/motioneye/motioneye.conf": [Errno 2] No such file...oneye.conf'
sept. 15 14:20:31 karmapi motioneye[4452]: failed!
sept. 15 14:20:31 karmapi systemd[1]: motioneye.service: control process exited, code=exited status=1
sept. 15 14:20:31 karmapi systemd[1]: Failed to start LSB: Start the motionEye server.
sept. 15 14:20:31 karmapi systemd[1]: Unit motioneye.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.

should i manually copy me motioneye.conf sample to /etc/motioneye ?

thanks for your help

ccrisan, 2015-09-15:
See https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian, step 4.

closed 2015-09-15 by ccrisan

villamarinella, 2015-12-01:
Starting motioneye (via systemctl): motioneye.service.
It started one time after installation.
But it couldnt use the raspi camera. The camera is working.
But now, after reboot its finished
And after 3 wrong trials with old versions I was happy the installation did run wo problems.

Klaus
2@sedara.net

villamarinella, 2015-12-01:
sorry, I forgot:

http://192.168.1.121:8765/