motioneye-project/motioneye #15: Allow more motion detection tuning options

hasbridge, 2015-09-16:
I am getting tons of false alerts from my outdoor cameras, due to tree shadows moving around in the wind. It would be nice to be able to specify a greater pixel change percentage than 20, as well as setting the noise level greater than 25%. Also adding support for the despeckle options would be nice.

ccrisan, 2015-09-17:
Motion detection mask is on the way.

ektus, 2015-12-26:
Is there any progress on the detection mask project? I've got a wide-angle outdoor camera that's triggering lots of false motion recordings due to movement of trees in the field of view. I would like to mask the trees away to greatly improve this situation.

DawsonLeBrown, 2016-06-01:
As Motion detection mask is already on the way, it's probably already been thought of or done, but it would be great if the current mask could be visualised through the camera preview - this way you will know that you have the mask set up correctly AND can confirm that the mask file you have created is covering the correct part of the image.
- btw, thanks for all your work on this.

ccrisan, 2016-10-04:
Added in 0.34.

closed 2016-10-04 by ccrisan