motioneye-project/motioneye #17: Error when trying to start

Gutz-Pilz, 2015-09-17:
```
root@pi-server:~# /etc/init.d/motioneye start
Starting motionEye server:Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2707, in <module>
    working_set.require(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 686, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 584, in resolve
    raise DistributionNotFound(req)
pkg_resources.DistributionNotFound: jinja2
 failed!
```

kristofejro, 2015-10-17:
I have something similar. Does anyone has idea how to solve the problem?

root@raspberrypi:/home/pi/motioneye# /etc/init.d/motioneye start
[....] Starting motionEye server:Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2707, in <module>
    working_set.require(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 686, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 584, in resolve
    raise DistributionNotFound(req)
pkg_resources.DistributionNotFound: certifi
 failed!

ccrisan, 2015-10-17:
That should be solved by a "sudo pip install certifi", although I have no idea why it isn't already present on your system. How did you install motionEye?

closed 2015-12-26 by ccrisan