motioneye-project/motioneye #18: Mr Dave's fork - motion not found on boot

electronicamystica, 2015-09-18:
I installed motionEye with Mr Dave's motion (I needed to do these steps for ffmpeg btw: https://github.com/ccrisan/motioneye/issues/11#issuecomment-138605394) and motioneye failed starting on boot:
           INFO: hello! this is motionEye server 0.27.2
           INFO: motion not installed
after logging in however 
            /etc/init.d/motioneye start
would have it start just fine. Setting the motion binary path in /etc/motioneye/motioneye.conf to
           /usr/local/bin/motion
solved that issue. 

I was just following the installation steps from the wiki on a clean raspbian install on a rasbpi B (Linux raspberrypi 4.1.6+ #810 PREEMPT )
These were the only issues, motioneye works like a charm, thanks!

ccrisan, 2015-09-18:
The path to /usr/local/bin is probably missing from the init script. I'll update it so that it works without additional tweaks.

ccrisan, 2015-09-18:
Fixed with the last commit.

closed 2015-09-18 by ccrisan

sve95, 2015-11-22:
Hello, i have tha same problem with motioneye  0.28.3

motioneye failed starting on boot
root@# /etc/init.d/motioneye status
[FAIL] motioneye is not running ... failed!

i check logfile 
root@# cat /var/log/motioneye.log
2015-11-22 12:29:35: [motioneye]     INFO: hello! this is motionEye server 0.28.3
2015-11-22 12:29:36: [motioneye]     INFO: motion not installed
2015-11-22 12:29:36: [motioneye]     INFO: ffmpeg not installed
2015-11-22 12:29:36: [motioneye]     INFO: v4l-utils not installed

motion installed in
root@# which motion
/usr/local/bin/motion

 init script contain 
PATH_BIN="/bin:/usr/bin:/sbin:/usr/sbin:/usr/local/bin:/usr/local/sbin"

when i start it manually
root@# /etc/init.d/motioneye start
[ ok ] Starting motionEye server:.

ccrisan, 2015-11-22:
@sve95 have you updated your init script (copied over the new version)? The init script doesn't get updated automatically with `pip install --upgrade motioneye`

sve95, 2015-11-22:
it clean install motioneye 0.28.3 on rpi.
I just follow installation Instructions and type 

pip install motioneye

i not modify init script

ccrisan, 2015-11-22:
Could you please rename `PATH_BIN` to `PATH` in `/etc/init.d/motioneye` see if it works?

sve95, 2015-11-22:
it works

/etc/init.d/motioneye status
[ ok ] motioneye is running.

thanks a lot

ccrisan, 2015-11-22:
You're welcome. I'll change in the script to reflect this fix.