motioneye-project/motioneye #21: Chrome doesn't propose to save login/password after login

PaulAnnekov, 2015-10-04:
After successful login as admin, Chrome does not propose to save login/password. I need to enter credentials again and again...

ccrisan, 2015-10-04:
Are you sure you haven't disabled credentials saving for your motionEye server? Check your chrome settings and if the problem still persists, please try it with another computer running Chrome, just so we can know it's an issue with motionEye and not with your Chrome installation.

PaulAnnekov, 2015-10-05:
@ccrisan it's working in FF (41.0.1) but not in Chrome (45.0.2454.101 (64-bit)). It's behind nginx.

BTW, I don't have any auth cookies and I need to re-enter login/password after page refresh. Doesn't motioneye uses auth cookies or it's my problem?

ccrisan, 2015-10-06:
I myself use motionEye behind nginx and it seems to work alright for me. What version of motionEye do you run by the way?

There's one cookie related to the authentication mechanism (the `username` cookie) which helps identifying the current role (*admin* vs *surveillance user*) but has no impact in remembering the credentials.

The login dialog contains a form with usual normal *username* and *password* fields, so browsers can detect it as a login form.

Could you please try accessing your motionEye directly (without passing through nginx) and see if the problem persists?

PaulAnnekov, 2015-10-06:
@ccrisan I have removed redirect from 80 to 443 in nginx and tested it. Passwords save in http works perfectly, but in https they are still not working. I tested it in Chrome from Windows and Linux.

I have found some interesting posts about https and form save, e.g. this one [Google Chrome does not offer to save password for https](https://productforums.google.com/forum/#!topic/chrome/jaCNKslaQY8).

ccrisan, 2015-10-06:
I wouldn't know what to say here. I also use HTTPS and my Linux Chrome offers to save my credentials without any issue.

PaulAnnekov, 2015-10-07:
@ccrisan do you have Chrome or Chromium? And what version?

ccrisan, 2015-10-07:
I use Chromium 45.0.2454.101 (64-bit) on Arch Linux. Do  you think the problem might be with Chrome and not with Chromium? I never thought they differed that much.

PaulAnnekov, 2015-10-08:
@ccrisan I have tested it a little bit in Chromium and that's what I have found. When you do not authorize your profile with Google Account then passwords save is working. But when you authorize your profile with Google Account it stops working.

ccrisan, 2015-10-09:
Interesting. I'll look into it.

PaulAnnekov, 2015-10-09:
@ccrisan today I have tested passwords save from different Google Account in Chrome on Windows. And it works too. I tested it from my Google Account on the same environment and it doesn't work.

So I think that something is wrong with my Google Account. Maybe it's because I have [encrypted all passwords with passphraze](https://support.google.com/chrome/answer/1181035?hl=en).

ccrisan, 2015-10-09:
Does credentials remembering work for other websites or services with your current Chrome configuration?

PaulAnnekov, 2015-10-12:
@ccrisan yes, it's working for github, bitbucket, facebook...

ccrisan, 2015-10-13:
Well then it's something to look into. I'll try to configure a Chrome profile with the indicated encrypted settings, so I can reproduce the problem.

PaulAnnekov, 2015-10-16:
Chrome is updated to 46.0.2490.71. Now it's working :).

ccrisan, 2015-10-17:
Great to hear that. I'll close the issue then and should the problem reappear, feel free to reopen it.

closed 2015-10-17 by ccrisan