motioneye-project/motioneye #23: How to update config using cli w/o motioneye restart

PaulAnnekov, 2015-10-10:
I need to update the motion detection option from cli when I detect needed mac addresses in my network. But I can't restart motioneye because I use docker. I need to update config w/0 motioneye restart. Do we have such feature?

ccrisan, 2015-10-11:
Nope, not yet. It's planned though, and it's going to be part of an API that can be used for scripting such behaviours.

PaulAnnekov, 2015-10-12:
@ccrisan I have decided to enable/disable motion detection using [motion API](http://www.lavrsen.dk/foswiki/bin/view/Motion/MotionHttpAPI#Detection_Commands). I will run curl to detection start/pause urls.

ccrisan, 2015-10-13:
motionEye might override that setting, so you might not get the expected result.

PaulAnnekov, 2015-10-13:
@ccrisan when will it do this? I have disabled the schedule in GUI.

ccrisan, 2015-10-14:
If working schedule is disabled, motionEye should left it untouched.

PaulAnnekov, 2015-10-16:
I'm trying to make requests to motion using `requests` module:
`requests.get('http://127.0.0.1:7999/1/detection/pause')`
And get the following response:
```
<html><head><title>Motion Unofficial-Git-2bd79c7</title></head>
<body>
<a href=/1/detection>&lt;&ndash; back</a><br><br>
<b>Thread 1</b> Detection paused
</body>
</html>
```
But I see that motion detection in motioneye GUI is still working and pictures/videos are created. I tried to enable and use `Working schedule` (it uses the same requests) and it's working w/o any problems. 

What's wrong with my request?

closed 2015-12-26 by ccrisan