motioneye-project/motioneye #27: failed to read settings, unknown configuration option

sinbrkatetete, 2015-10-17:
Total noob here, sorry for possibly using the wrong channel to seek help and perhaps report an issue. Please do point me in the right direction if it's not too much trouble.

I'm trying to set up an EC2 Ubuntu server instance and motioneye on it, to control two of my raspberry pis running motionpie (since one of them is a B version, I'd like to try and set it up as a fast network cam and do all the motion work on the EC2 instance). I've followed your installation instructions for Ubuntu and got to the starting of the motionEye server, but the following pops up:
"CRITICAL:root:failed to read settings from "/etc/motioneye/motioneye.conf": unknown configuration option: motion:control_localhost"

I'm using your sample conf as the starting conf file. http://pastebin.com/0dW0G1SS

ccrisan, 2015-10-19:
What version of motionEye do you use?

sinbrkatetete, 2015-10-19:
Hey. Thanks for taking the time to answer. Even bigger thanks for taking the time to bring us these great tools. I'm using version 0.28.1
I've commented the setting out and now I'm testing what does and what does not work.... For now, I've managed to connect the cameras as remote MotionEye cameras. It seems the my local upload speed poses the biggest problem.

pengocha, 2015-10-19:
Got the same here today, sorry for the double post.

ccrisan, 2015-10-19:
It's a known bug, recently fixed. The fix will be available in 0.28.2. Until then, please just comment out "motion_control_localhost" in your motioneye.conf.

closed 2015-10-19 by ccrisan

pengocha, 2015-10-19:
I put '#' before motion_control_localhost true, get message back saying: 'CRITICAL:root:failed to read settings from "/etc/motioneye/motioneye.conf": invalid configuration line: '

ccrisan, 2015-10-19:
Can I see your config file?

pengocha, 2015-10-19:
Sure: http://pastebin.com/zv0iuEvc

ccrisan, 2015-10-19:
Your config file seems alright. Are you sure you don't have any hard-to-spot stray character somewhere in the file?

pengocha, 2015-10-19:
Not that i know or see. For tonight im done. I'll try tommorow to delete the entire file and replace it with the one pasted by sinbrkatetete earlier. Maybe that will help. I'll let you know.

pengocha, 2015-10-20:
You where right, on top of motioneye.conf was an lost $. All working again. Thanks so much !!