motioneye-project/motioneye #28: MotionEye always reboot

Cyril25, 2015-10-18:
I don't know why but since this afternoon, my raspberry pi with MotionEyeOs is always rebooting ...
It's says that eth0 has no IP address but the respberry pi is connected at my network ...
Thanks
[image]

sinbrkatetete, 2015-10-18:
Noob here, so I'm sorry if I'm pointing out the obvious. My intention is only to try and help until someone more knowledgeable comes along. You could try setting the "network link watch" under "expert settings" to off, or try prolonging the timeout, just to see if everything else works ok. If it does, try setting the fixed ip under "network" and setting the "network link watch" back on. If everything still works ok,, then try connecting the Pi through wireless if possible, but with dhcp assigned IP and "network link watch" on. This way, as far as I know (and I know very little), you can narrow the culprit to the software/hardware part that's giving you trouble. If it works with wireless, it could be your ethernet adapter (software) malfunctioning. If it doesn't, it could be your dhcp server not asigning the IP... Did you try rebooting the router?

Cyril25, 2015-10-18:
Hi, thank you for your help ! :)

 

I just tried to prolonging the timeout to 60seconds, to set the network link watch to off, but nothing works.

I already have an Fixed IP.

For the moment I’m waiting for my wifi dongle, so I can’t try a Wifi connexion.

I already tried to reboot the router, but nothing better …

 

I tried to disconnect one of the 3 webcam I have on the motionEyeOs (2 USB and 1 IP), and after, my raspberry was OK,

When I tried to reconnect the 3rd webcam, the raspberry reboot everytime …

I tried to delete and re-add the camera, and I noticed that if the video resolution is 320*240, all goes well, and if I put 640*480 (the best resolution proposed), here there is the problem !

 

Sorry if I’m not really clear but my English is not as good as I would like…

I think that is less a network problem, than a webcam, or video resolution problem, but I really don’t know …

If you, or somebody else can help me to understand and resolve the problem, it would be really great,

Thank you,

Cyril

ccrisan, 2015-10-19:
You might have a power issue. Please try using a powered USB hub.

Cyril25, 2015-10-21:
I don't have any USB hub on my raspberry pi, I don't understand how I can do that.
Since 3 days I don't have any issues, any reboot, so the video resolution of the USB camera is maybe the reason of my problem ...Thank you for your help ! :)

duc-nguyen-locm, 2015-11-14:
what ccrisan saying is that the power on your RPi is not enough to run all the cameras you plugged in.  Use an external USB hub with its own power supply to connect your cameras.

Cyril25, 2015-11-16:
Oh ok ! Thank you for this precision, I will try that ASAP ! Have a good day.

closed 2015-11-16 by Cyril25

mastix, 2015-11-22:
Having the same (or at least a similar) problem here... Installed motionEyeOS on my Rasp2 today (latest downloadable version 1103).

eth0 is attached directly to my router.
No cameras or what so ever attached to my Pi.

During boot it keeps trying to get an IP from my router, but fails after some time (all my other raspberries are getting an IP and there are still free IPs left).

Then after some time it's telling me that eth0 didn't get an IP address and reboots the device... (I can even see the "Welcome to motionEye" message), but since eth0 didn't get an IP address, it just reboots.

Other Raspberries (running raspbian jessie) attach to the network just fine... even when using the same cable & port.

Any help would be appreciated. Otherwise I would have to run Rasbpian with MotionEye on top, but your slim solution is just so much cooler. :)

Regards,

Sascha

ccrisan, 2015-11-22:
If you connect any other regular device to your router, approximately how long do you have to wait until it gets an IP address?

mastix, 2015-11-22:
It's a matter of a couple of seconds... So every other device works perfectly fine. I could install raspbian later to proof that eth0 works fine normally.

ccrisan, 2015-11-22:
Please do, as you're the only one so far experiencing problems with DHCP, so there might be a problem with your device or with the ethernet connection. Keep me posted.

mastix, 2015-11-22:
Just installed raspbian via netinst (https://github.com/debian-pi/raspbian-ua-netinst) and seconds later it got an IP address and downloaded all packages. Now I'm connected via SSH to my new install.

mastix, 2015-11-22:
Ok... update: I've just tried to install motionEyeOS a third time (after flashing raspbian in between)... now it picked up an IP address... :) Don't ask me why... maybe netinst flashed a new firmware or whatever... nevertheless it seems to work now! :)

ccrisan, 2015-11-22:
Alright, that's great news.