motioneye-project/motioneye #31: Is it possible to force motion trigger?

lwrbakro, 2015-10-20:
I have a PIR sensor connected to my Pi.

Is it possible to force motion trigger outside motioneye?

ccrisan, 2015-10-20:
Not yet.

mvanbreeden, 2016-01-13:
It does seem from motion source that one is able to send motion a SIGALRM to trigger capture...

Daniel-1276, 2016-01-22:
Yes, and SIGUSR1 seems to start movie capture ... SIGTERM/SIGQUIT/SIGINT seem to close the video stream afterwards. I would really like to see PIR motion sensor feature within motioneye, as in many cases it is more reliable if you want to secure places against burglars, as the emit heat and can be detected more reliable maybe.

gunther-o, 2016-09-14:
I can't  create a motion trigger with SIGALRM (tested on motioneyeos 20160801)
Is there an example or docu available ?
(I am highly interested in PIR motion trigger)

melashkov, 2016-10-17:
Is there is anything I can help with, can somebody point where to look?

peterwtux, 2017-07-12:
If you enable webcontrol in motion.conf, you can make HTTP requests to set emulate_motion=on to make motion behave as if it detected motion, and then set emulate_motion=off when your PIR sensor reports no movement. https://htmlpreview.github.io/?https://github.com/Motion-Project/motion/blob/master/motion_guide.html#emulate_motion

wb666greene, 2017-09-21:
I'm of the opinion you want the logical AND of video (image) based motion detection and PIR based motion detection.  PIR picks up cats, squirrels, etc. along with fast moving clouds on a sunny day, along with temperature changes as the sun rises and sets, etc.  Another issue with PIR is in our hot climate, the PIR can be essentially blind to people when the ambient temperature is about body temperature -- more common here than anyone really likes :)

The most effective use of either is in locations where you control the illumination of the field of view and there are no "rapid" temperature changes.

It would be a major feature if added to motion.  I've looked at the source code, but without guidance I've no idea where the motion is detected in the "pre-motion" buffers and what triggers writing of the pre-motion images and stops recording of subsequent frames.

peterwtux, 2017-09-21:
@wb666greene, what if you use an external utility to monitor your PIR sensor(s) and change the "threshold" value via motion's webcontrol interface? Set the default threshold high to avoid image-based false alarms, and have your PIR monitoring utility decrease motion's threshold when motion is detected. If you really want "and" logic, just set the default threshold impossibly high.

wb666greene, 2017-09-21:
@peterwtux, I think you are over estimating the speed at which the motion-daemon reacts to changes over the web interface, also changes to thresholds require overwriting the pre-motion buffers with new images.

I've used motion and zoneminder for a long time, motion has adapted to higher resolution much better than zoneminder has, and I have been happy with cameras placed where I mostly control the lighting in the field of view.  But I'm looking to  cover the perimeter for "early warning" alerts and so far all image/video and PIR motion detection have had an unacceptable false alarm rate.  YMMV.

The ultimate solution I'm looking at is "deep learning" image analysis using new tools like tensorflow, but the learning curve is substantial!  But motioneyes will give me a good source of images both true and false for analysis.

xtalker, 2017-11-02:
I got this command to work:

curl http://my.motioneye.com:7999/2/config/set?emulate_motion=on (or off)

But the first 10 secs of every video is just blackness, after that normal video?   Tried on the native RasPi camera and an IP camera as well with same results.

Anyone else seen this and found a setting or fix?

vganyn, 2017-11-02:
From web interface i get normal video

flabombarda, 2017-12-11:
Guys, 

I see this is an old request, and very useful, and is on the roadmap since 2015, is there any chance we have it in the upcoming releases ?

Many thanks

dezral, 2017-12-22:
+1 

just testet i can't access picam  if motion/eye is running so i can't use a python script with the pir sensor... so some kinder api that triggers motion recording in motioneye or access to gpio

FAASOFT, 2018-03-05:
to xtalker - 
I'm a beginner: could you please explain me how/where send the command? May be by a bash script under the PIR action? Can this be done on MotioneyeOS also? Any example would be appreciated

xtalker, 2018-03-05:
Hi FAASOFT, your motion eye web server should accept this command (sent from a bash script with `curl` or just from a browser). Don't forget to change `my.motioneye.com` to whatever your motioneye server name is.

http://my.motioneye.com:7999/2/config/set?emulate_motion=on (or off)

FAASOFT, 2018-03-05:
Hi Bob,

thank you for the prompt answer.
I have Motioneye running on a MotioneyeOS installation on Raspberry PI Zero W;
I can see the cam image at 192.168.2.101. If during it's normal operation I send by a browser the command http://192.168.2.101:7999/2/config/set?emulate_motion=on the result is ERR_CONNECTION_REFUSED.
My be the 7999 port is not open? Where I'm wrong?

Angelo

peterwtux, 2018-03-05:
Check your motion configuration. The default is for the control web service to be bound to localhost (webcontrol_localhost). The control web service is ripe for Cross Site Request Forgery attacks, so it's really a good idea to limit it to localhost, and only run motion on a single-user system. You might be able to use iptables owner rules to better secure the control web service if you have untrusted users or processes running on the same host as motion, but I haven't tried that.

jasaw, 2018-03-05:
@FAASOFT Make sure you have `webcontrol_parms 2` in your motion.conf file.
Also make sure you have the correct camera index. The number after port 7999 is the camera index.
`curl "http://localhost:7999/1/config/set?emulate_motion=1"`

schnabelnator, 2018-08-20:
Hi, i want to do the same thing, trigger motion by an external sensor connected to the gpio. Can anyone point me into the rigth direction on how to create a script that watches the gpio state and calls the curl command if the state changes? I don't know my way around motion os yet...

Where would i save my script and how would it start?

Thank you for your help!

AStripe, 2019-02-26:
> 
> 
> I have a PIR sensor connected to my Pi.
> 
> Is it possible to force motion trigger outside motioneye?

@lwrbakro did you find a way to use an external motion sensor? I'd like to make a camera using the ModMyPi case that has a PID Motion sensor

MichaIng, 2026-10-21:
With #2691 eventstart/eventend actions have been added. It works similar to `http://127.0.0.1:7999/1/config/set?emulate_motion=1`, but calls `http://127.0.0.1:7999/1/action/eventstart` and `eventend` resp. This starts/stops the motion event immediately, without event gap etc and does not require `webcontrol_parms 2`. And since it is made available through the motionEye API, it can be called from other hosts, who cannot/should not reach the motion webcontrol port directly.

Note that, since motionEye 0.44.0, a separate login call is needed: https://github.com/motioneye-project/motioneye/releases/tag/0.44.0

closed 2026-10-21 by MichaIng