motioneye-project/motioneye #32: update failed from 0.27.2 to 0.28.2

karma007, 2015-10-21:
Hello,

i have update using  pip install --upgrade motioneye (as root) in my RPI2.
Now motioneye is broken and the version shows 0.27.2 unless the message "Successfully installed motioneye-0.28.2 pillow-3.0.0"

[image]

thanks

karma007, 2015-10-21:
pip install --upgrade motioneye==0.27.2.post2

-> works again but pip install --upgrade motioneye failed again to work

ccrisan, 2015-10-21:
What is the output of the command `pip install --upgrade motioneye`?

karma007, 2015-10-21:
Collecting motioneye
Requirement already up-to-date: pycurl in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: tornado>=3.1 in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: pillow in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: jinja2 in /usr/local/lib/python2.7/dist-packages (from motioneye)
Requirement already up-to-date: certifi in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Requirement already up-to-date: backports.ssl-match-hostname in /usr/local/lib/python2.7/dist-packages (from tornado>=3.1->motioneye)
Requirement already up-to-date: MarkupSafe in /usr/local/lib/python2.7/dist-packages (from jinja2->motioneye)
Installing collected packages: motioneye
  Found existing installation: motioneye 0.27.2
    Uninstalling motioneye-0.27.2:
      Successfully uninstalled motioneye-0.27.2
Successfully installed motioneye-0.28.2

ccrisan, 2015-10-21:
Upgrading looks alright. And after upgrading, can you please send me the motioneye.log file? (usually found at /var/log/motioneye.log)?

karma007, 2015-10-21:
here it is
https://dl.dropboxusercontent.com/u/18722948/motioneye.log
note i had an error at first update because i forgot "sudo"

karma007, 2015-10-21:
here it is : https://dl.dropboxusercontent.com/u/18722948/motioneye2.log

ccrisan, 2015-10-21:
Wait what? Do you use motionEyeOS? If you do, you must not upgrade your motionEye manually. Instead wait for the next release of motionEyeOS.

ccrisan, 2015-10-21:
Now you changed the URL of the log file and it appears to be a totally different log file. I'm confused. What computer and what OS are you using? How did you install motionEye?

karma007, 2015-10-21:
sorry, i uploaded first a wrong log (from motioneyeos, i have 2 rpis). I installed motioneye on raspbian using your guide. thanks

ccrisan, 2015-10-21:
What does `cat /usr/lib/python2.7/site-packages/motioneye/__init__.py` say? If the file does not exist, what does `cat /usr/local/lib/python2.7/site-packages/motioneye/__init__.py` say?

karma007, 2015-10-21:
cat /usr/lib/python2.7/site-packages/motioneye/__init__.py -> file does not exist
cat /usr/local/lib/python2.7/site-packages/motioneye/__init__.py -> file does not exist

ccrisan, 2015-10-21:
It's `__init.py__`, not `init.py`.

karma007, 2015-10-21:
yes, 
/usr/lib/python2.7/site-package does not exist
/usr/local/lib/python2.7/site-packages is empty

ccrisan, 2015-10-21:
Alright, then you must have `/usr/lib/python2.7/dist-packages/motioneye/__init__.py`.

karma007, 2015-10-21:
i don't have any motioneye folder on /usr/lib/python2.7/dist-packages
but i have __init__.py in usr/local/lib/python2.7/dist-packages/motioneye/

cat /usr/local/lib/python2.7/dist-packages/motioneye/__init__.py : 

VERSION = "0.28.2"

i noticed i also have motioneye-0.28.2.dist-info folder in /usr/local/lib/python2.7/dist-packages

ccrisan, 2015-10-21:
Then everything seems to be fine. By the way have you restarted motionEye after upgrade (`/etc/init.d/motioneye restart`)?

karma007, 2015-10-21:
No. but here is the result : 
sudo /etc/init.d/motioneye restart
[....] Restarting motioneye (via systemctl): motioneye.serviceJob for motioneye.service failed. See 'systemctl status motioneye.service' and 'journalctl -xn' for details.

And : sudo systemctl status motioneye.service
● motioneye.service - LSB: Start the motionEye server
   Loaded: loaded (/etc/init.d/motioneye)
   Active: failed (Result: exit-code) since mer. 2015-10-21 15:38:42 CEST; 1min 38s ago
  Process: 21240 ExecStop=/etc/init.d/motioneye stop (code=exited, status=0/SUCCESS)
  Process: 21578 ExecStart=/etc/init.d/motioneye start (code=exited, status=1/FAILURE)

oct. 21 15:38:42 karmapi motioneye[21578]: Starting motionEye server:CRITICAL:root:failed to read settings from "/etc/motioneye/motioneye.con...atic_url
oct. 21 15:38:42 karmapi motioneye[21578]: failed!
oct. 21 15:38:42 karmapi systemd[1]: motioneye.service: control process exited, code=exited status=1
oct. 21 15:38:42 karmapi systemd[1]: Failed to start LSB: Start the motionEye server.
oct. 21 15:38:42 karmapi systemd[1]: Unit motioneye.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.

ccrisan, 2015-10-21:
Well not restarting it was the cause of your issue. Now the problem is that between 0.27 and 0.28 the settings have changed a bit. If you haven't touched your motioneye.conf, please remove it and copy the sample file over (see the install instructions for that).

closed 2015-10-21 by ccrisan

karma007, 2015-10-21:
And : journalctl -xn
-- Logs begin at lun. 2015-10-19 12:27:01 CEST, end at mer. 2015-10-21 15:42:04 CEST. --
oct. 21 15:42:04 karmapi motioneye[22530]: Starting motionEye server:CRITICAL:root:failed to read settings from "/etc/motioneye/motioneye.conf": unknown
oct. 21 15:42:04 karmapi sudo[22539]: pi : TTY=unknown ; PWD=/media/Data/Sauvegarde/XBMC/kankupicam ; USER=root ; COMMAND=/bin/rm -r /media/Data/Sauvega
oct. 21 15:42:04 karmapi motioneye[22530]: failed!
oct. 21 15:42:04 karmapi systemd[1]: motioneye.service: control process exited, code=exited status=1
oct. 21 15:42:04 karmapi systemd[1]: Failed to start LSB: Start the motionEye server.
-- Subject: L'unité (unit) motioneye.service a échoué
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- L'unité (unit) motioneye.service a échoué, avec le résultat failed.
oct. 21 15:42:04 karmapi systemd[1]: Unit motioneye.service entered failed state.
oct. 21 15:42:04 karmapi sudo[22521]: pam_unix(sudo:session): session closed for user root
oct. 21 15:42:04 karmapi sudo[22539]: pam_unix(sudo:session): session opened for user root by (uid=0)
oct. 21 15:42:04 karmapi sudo[22539]: pam_unix(sudo:session): session closed for user root
oct. 21 15:42:04 karmapi CRON[22458]: pam_unix(cron:session): session closed for user pi

karma007, 2015-10-21:
failed to read settings from "/etc/motioneye/motioneye.conf": unknown
this file is rw-r--r--

ccrisan, 2015-10-21:
As I said, use the new sample file that comes with 0.28.2.

karma007, 2015-10-21:
with this command line ?
sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
is that file the 0.28.2 sample ?
thanks

ccrisan, 2015-10-21:
Yeap :)

ccrisan, 2015-10-21:
And don't forget to restart it afterwards.

karma007, 2015-10-21:
thanks, it works.
Should i copy this file again for next releases/updates ?

ccrisan, 2015-10-21:
Normally not, unless you get errors related to the settings after upgrading.