motioneye-project/motioneye #33: Still images are recorded but not movies suring motion detection

karma007, 2015-10-21:
Hello, i noticed this problem in version 0.27.2 and now in 0.28.2.
Pictures are saved, but movies are not.

thanks
[image]

ook, 2015-10-21:
Hi. Have you installed required tools for building movies? (depend of your architecture, should be avconv or ffmpeg IIRW)

ccrisan, 2015-10-21:
`/var/log/motion.log` should say more about this problem.

karma007, 2015-10-21:
here is the log (i'm on RPI2) : 
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_output_movies"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_variable_bitrate"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_video_codec"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "ffmpeg_bps"
[0] [ALR] [ALL] conf_cmdparse: Unknown config option "movie_filename"
[0] [NTC] [ALL] motion_startup: Using default log level (NTC) (6)
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-d71cebf Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (NTC)
[0] [NTC] [ALL] main: Thread 1 is from thread-1.conf
[0] [NTC] [ALL] main: Thread 1 is device: http://192.168.1.23:8081/ input -1
[0] [NTC] [ALL] main: Stream port 8081
[0] [NTC] [ALL] main: Waiting for threads to finish, pid: 26584
[1] [NTC] [ALL] motion_init: Thread 1 started , motion detection Enabled
[0] [NTC] [STR] httpd_run: motion-httpd testing : IPV4 addr: 127.0.0.1 port: 7999
[0] [NTC] [STR] httpd_run: motion-httpd Bound : IPV4 addr: 127.0.0.1 port: 7999
[0] [NTC] [STR] httpd_run: motion-httpd/Unofficial-Git-d71cebf running, accepting connections
[0] [NTC] [STR] httpd_run: motion-httpd: waiting for data on 127.0.0.1 port TCP 7999
[1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (http://192.168.1.23:8081/)
[1] [NTC] [NET] netcam_read_first_header: Found Conn: close header ('Connection: close')
[1] [NTC] [NET] netcam_setup_html: connected, going on to read image.
[1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[1] [ALR] [NET] netcam_handler_loop: Camera handler thread [2] started
[1] [NTC] [STR] http_bindsock: motion-stream testing : IPV4 addr: 127.0.0.1 port: 8081
[1] [NTC] [STR] http_bindsock: motion-stream Bound : IPV4 addr: 127.0.0.1 port: 8081
[1] [NTC] [ALL] motion_init: Started motion-stream server in port 8081 auth Disabled
[1] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 3 items
[1] [WRN] [NET] netcam_next: called with no data in buffer
[0] [NTC] [STR] httpd_run: motion-httpd - Read from client

I have followed this procedure : https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

whereis ffmpeg
ffmpeg: /usr/bin/ffmpeg /usr/share/man/man1/ffmpeg.1.gz

ccrisan, 2015-10-22:
Your motion has no ffmpeg support. How did you install motion?

karma007, 2015-10-22:
i think i used "apt-get install motion ffmpeg"
but, as i used motioneye old version, i'm not sure.
More over, i think my motion file in /usr/bin/motion is "motion--mmal" (as i've got a motion.old too).
Shoud il replace this file with this one : https://github.com/ccrisan/motioneye/wiki/motion-binaries/motion-mrdave-raspbian ?

thanks

ccrisan, 2015-10-22:
Yes, you should.

karma007, 2015-10-22:
i have downloaded and chmod 0755/usr/bin/motion
now movies works again
thanks

ccrisan, 2015-10-22:
Great. I'll close the issue then.

closed 2015-10-22 by ccrisan

2ra3, 2017-04-07:
thread-1 is not being procesed. what am i doing wrong?

ccrisan, 2017-04-08:
@2ra3 please open a specific issue and give us more details (such as how is it not being processed and some log/config files).