motioneye-project/motioneye #34: Can't install!

parkerste, 2015-10-23:
OK I admit I'm a newbie... but... Ive manages to get my pi running OpenVPN and an AirTunes server... I even rebuilt it with motionPie... and thought... why not get it to do all three!

So I went back to a basic rasbian install.
I followed the wiki and when I downloaded git clone https://bitbucket.org/ccrisan/motioneye.git
the motioneye.pl file is empty and there is no setup_example config file

Am I missing something or am I just too much of a newbie?

Thanks
Steve

sinbrkatetete, 2015-10-23:
Fellow noob here. Did you try installing motioneye by following the instructions at https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian ?

ccrisan, 2015-10-23:
`motioneye.pl`? `setup_example`? Are you sure we're talking about the same motionEye?

parkerste, 2015-10-23:
Is there a different motionEye?  I think I just got tired and confused.
I'll try working through the link for a raspbian install tomorrow.   It
seems to ensure I get the right version of tornado

I meant settings_default.py which needs to be copied to settings.py

I start off with a clean install of 2015-05-05-raspbian-wheezy.img on my sd
card.
expand the file system, change the password...

then I download the script rename it to something sensible and use sh to
run it... all looks good at the end but there's no browser access on port
8765

Here's the scripts I used... any thoughts?

clear
echo 'Starting Installation...'
sleep 2
cd /home/pi
sudo apt-get update && apt-get upgrade -y
sudo apt-get install motion ffmpeg v4l-utils python-dev python-pip
libssl-dev libjpeg-dev libcurl4-openssl-dev -y
sudo pip install tornado==4.1 jinja2 pillow pycurl
git clone https://bitbucket.org/ccrisan/motioneye.git
cd motioneye/
cp settings_default.py settings.py
mkdir run
cd /etc
sudo rm -rf rc.local
sudo wget https://goo.gl/69kvdv
sudo mv 69kvdv rc.local
sudo chmod +x rc.local
sudo chmod 777 rc.local
cd /home/pi/
clear
echo 'Installation Complete! Preparing to REBOOT :)'
sleep 5
sudo reboot

#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

sudo -u pi sh -c '/home/pi/motioneye/motioneye.py >
/home/pi/motioneye/run/motioneye.log 2>&1' &

exit 0

parkerste, 2015-10-24:
So I started fresh today... running the update and upgrade separately to
check each ran ok,

Ignore the previous script... I followed the Install On Raspbian to the
letter...
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

apt-get install motion ffmpeg

   1.

   apt-get install python-pip python-dev libssl-dev
libcurl4-openssl-dev libjpeg-dev

   1.

   pip install motioneye

The pip install motioned looked far more promising... but...
after successfully creating

mkdir -p /etc/motioneye

the copy fails

   1.

   cp /usr/local/share/motioneye/extra/motioneye.conf.sample
/etc/motioneye/motioneye.conf

There is no /motioneye/extra/ directory in /usr/local ...so no
motioneye.conf.sample to copy and /etc/motioneye is therefore empty.

Could this have anything to do with my changing the default password?  I
always do that!

Any thoughts?

*pi@raspberrypi* *~ $* pip install motioneye

Downloading/unpacking motioneye

  Downloading motioneye-0.28.2.tar.gz (524Kb): 524Kb downloaded

  Running setup.py egg_info for package motioneye

Downloading/unpacking tornado>=3.1 (from motioneye)

  Downloading tornado-4.3b1.tar.gz (453Kb): 453Kb downloaded

  Running setup.py egg_info for package tornado

    no previously-included directories found matching 'docs/build'

    warning: no files found matching 'tornado/test/README'

Downloading/unpacking jinja2 (from motioneye)

  Downloading Jinja2-2.8.tar.gz (357Kb): 357Kb downloaded

  Running setup.py egg_info for package jinja2

    warning: no files found matching 'run-tests.py'

    warning: no files found matching '*' under directory 'custom_fixers'

    warning: no files found matching '*' under directory
'jinja2/testsuite/res'

    warning: no previously-included files matching '*' found under
directory 'docs/_build'

    warning: no previously-included files matching '*.pyc' found under
directory 'jinja2'

    warning: no previously-included files matching '*.pyc' found under
directory 'docs'

    warning: no previously-included files matching '*.pyo' found under
directory 'jinja2'

    warning: no previously-included files matching '*.pyo' found under
directory 'docs'

Downloading/unpacking pillow (from motioneye)

  Downloading Pillow-3.0.0.tar.gz (9.6Mb): 9.6Mb downloaded

  Running setup.py egg_info for package pillow

Downloading/unpacking pycurl (from motioneye)

  Downloading pycurl-7.19.5.1.tar.gz (142Kb): 142Kb downloaded

  Running setup.py egg_info for package pycurl

    Using curl-config (libcurl 7.26.0)

Downloading/unpacking backports.ssl-match-hostname (from
tornado>=3.1->motioneye)

  Downloading backports.ssl_match_hostname-3.4.0.2.tar.gz

  Running setup.py egg_info for package backports.ssl-match-hostname

Downloading/unpacking singledispatch (from tornado>=3.1->motioneye)

  Downloading singledispatch-3.4.0.3.tar.gz

  Running setup.py egg_info for package singledispatch

Downloading/unpacking certifi (from tornado>=3.1->motioneye)

  Downloading certifi-2015.9.6.2.tar.gz (368Kb): 368Kb downloaded

  Running setup.py egg_info for package certifi

Downloading/unpacking backports-abc>=0.4 (from tornado>=3.1->motioneye)

  Downloading backports_abc-0.4.tar.gz

  Running setup.py egg_info for package backports-abc

Downloading/unpacking MarkupSafe (from jinja2->motioneye)

  Downloading MarkupSafe-0.23.tar.gz

  Running setup.py egg_info for package MarkupSafe

Downloading/unpacking six (from singledispatch->tornado>=3.1->motioneye)

  Downloading six-1.10.0.tar.gz

  Running setup.py egg_info for package six

    no previously-included directories found matching 'documentation/_build'

Installing collected packages: motioneye, tornado, jinja2, pillow, pycurl,
backports.ssl-match-hostname, singledispatch, certifi, backports-abc,
MarkupSafe, six

  Running setup.py install for motioneye

    error: could not create
'/usr/local/lib/python2.7/dist-packages/motioneye': Permission denied

    Complete output from command /usr/bin/python -c "import
setuptools;__file__='/home/pi/build/motioneye/setup.py';exec(compile(open(__file__).read().replace('\r\n',
'\n'), __file__, 'exec'))" install --single-version-externally-managed
--record /tmp/pip-rpYzI7-record/install-record.txt:

    running install

running build

running build_py

creating build

creating build/lib.linux-armv7l-2.7

<snip of all intervining OK msgs>

copying motioneye/templates/main.html ->
build/lib.linux-armv7l-2.7/motioneye/templates

running install_lib

creating /usr/local/lib/python2.7/dist-packages/motioneye

error: could not create '/usr/local/lib/python2.7/dist-packages/motioneye':
Permission denied

----------------------------------------

Command /usr/bin/python -c "import
setuptools;__file__='/home/pi/build/motioneye/setup.py';exec(compile(open(__file__).read().replace('\r\n',
'\n'), __file__, 'exec'))" install --single-version-externally-managed
--record /tmp/pip-rpYzI7-record/install-record.txt failed with error code 1
in /home/pi/build/motioneye

Storing complete log in /home/pi/.pip/pip.log

*pi@raspberrypi* *~ $* mkdir -p /etc/motioneye

mkdir: cannot create directory `/etc/motioneye': Permission denied

*pi@raspberrypi* *~ $* sudo mkdir -p /etc/motioneye

*pi@raspberrypi* *~ $* sudo cp
/usr/local/share/motioneye/extra/motioneye.conf.sample
/etc/motioneye/motioneye.conf

cp: cannot stat `/usr/local/share/motioneye/extra/motioneye.conf.sample':
No such file or directory

*pi@raspberrypi* *~ $* sudo cp
/usr/local/share/motioneye/extra/motioneye.conf.sample
/etc/motioneye/motioneye.conf

cp: cannot stat `/usr/local/share/motioneye/extra/motioneye.conf.sample':
No such file or directory

parkerste, 2015-10-24:
I am a proper NOOB!!!!!....
It was all down to permissions and (lack of) sudo... I got it working.
Brilliant software, just with I wasn't so thick ;)

ccrisan, 2015-10-26:
Where did you find this script? The `settings_default.py` file is part of an older motionEye version. The instructions @sinbrkatetete gave you should walk you through all the steps.

parkerste, 2015-10-26:
I can't recall where I got this script from, I had been trying, and failing
with motion for some time before I discovered motioneye, and so could have
had half installations left there with bits of script not run because of
permissions...  So I started with a clean install.

The point is https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
works!

The only thing I would add is run everything as root, so add...
sudo -s

Otherwise it's brilliant!  Thank you so much

I now have openvpn running alongside so I can securely access motioneye.
That was quite a struggle, especially the dynamic DNS, and although it's
working, I'm not 100% sure I have got ddclient set up properly.

I still consider myself a NOOB, but I'm getting the hang of things now.
Perhaps I should have bought a book.

Hmmm now what else shall I get my pi to do?

Thanks again.
Steve

closed 2015-10-26 by ccrisan