motioneye-project/motioneye #35: Missing Icons Webinterface

cryptoluks, 2015-10-29:
Hi!

I installed Motioneye like in the Wiki and all worked fine for some days.
But now, all images (like buttons, etc) are missing in the webinterface.

What I have done already:

- removed motioneye (pip uninstall) and all dependencies
- deleted pip cache
- deleted config files
- reinstalled --> same

Before it got 'broke' I worked on the nginx config to access motioneye via ssl.

- Using Archlinux 4.2.4
- running motioneye with root, so there should be no permission problem to access img folder
- same result with port 8765 / nginx proxy
- same result on different devices

--> I discovered, that PIL and pillow where both installed. But they are in conflict? I uninstalled PIL, reinstalled everything.. but same result..

Thank you!

cryptoluks, 2015-10-29:
[image]

ccrisan, 2015-10-29:
Does `pacman -S mime-types` solve the problem?

cryptoluks, 2015-10-29:
Yes!! Thank you very much! :+1:

closed 2015-10-29 by cryptoluks

ccrisan, 2015-10-29:
I'll update the wiki page accordingly.