motioneye-project/motioneye #37: Interval and motion detection together

lwrbakro, 2015-11-05:
It would be great to have interval and motion detection (with work schedule)
still image recording.

I know it is possible to write cron to change settings time to time.
But with custom (gpio) trigger event it make sense.

jamox, 2016-01-06:
+1 ability to do motion detection and create timelapse would be awesome. In my case I have camera watching over some land I have, mainly to detect animals walking around. It would be neat to be able to take snapshots and then manually generate timelapse of those.

lwrbakro, 2016-01-07:
You can do snapshots with x FPS triggered by motion on the picture.
After that, if you see these pictures there is a Timelapse button

ozett, 2017-06-07:
> You can do snapshots with x FPS triggered by motion on the picture.
After that, if you see these pictures there is a Timelapse button

do you mean, i have to activate the picture option, 
than wait, that there are pictures,
than go into the picture browser,
than there is an option to get a movie from that pictures?

i as this, because a cannot find a way to set up timelapse movies in motion-eye,
like i saw this on motion itself.

ozett, 2017-06-07:
i found this comment, but nothing in the wiki.

https://github.com/ccrisan/motioneye/issues/464#issuecomment-283365174

seems like this is telling how that should work. i will try taking pictures and see in the picture-browser if i can get a movie tomorrow.

surak, 2018-07-30:
The other comment is in a closed issue. I would also like to be able to take pictures periodically PLUS having motion detection.

degerrit, 2019-04-24:
I'm currently migrating my motion+home made shell scripts setup to motioneye (awesome s/w, btw!)

Most of my config has been ported over, but this issue bit me too - I have to choose betwen "Motion triggered" Still Image capture mode, or Interval snapshots, whereas in the original motion software they are not mutually exclusive options at all! So it seems a UI issue mainly.

I tried adding my old config line `snapshot_interval 1800` to "Extra Motion Options", but MotionEye is too clever and assumes that means I want to disable Motion Triggered capture.

What would work for me is to just be able to add the "Extra Motion Options" line and have MotionEye leave it alone rather than interpret my intentions (i.e. selecting both options doesn't need to be in the UI per se)

mikavaliviita, 2019-05-15:
Also a +1 for an option to have both motion triggering and interval shooting available simultaneously. For example, you could have normal motion triggering available outside working hours but also take some images every hour always.

@ccrisan , what do you say?

Thanks,

alexw22, 2021-01-26:
+1 Really would love to have this feature too! Motion for when things are happening together with a timelapse video for every day would be awesome!

degerrit, 2021-02-01:
My workaround has been a cronjob `*/5 * * * *` to make the snapshot; something like this:

```
#!/bin/bash
FILENAME="/home/_webcam/cam1/"$(date +'%Y-%m-%d')/$(date +'%Y%m%d-%H%M-%S--snapshot-0').jpg
/usr/bin/curl --max-time 10 --output $FILENAME http://127.0.0.1:8765/picture/2/current >/tmp/curl-snapshot.log 2>&1

# then, this is the same script I used to have in motion.conf for on_picture_save (it may not be necessary in everyone's config):
/usr/local/bin/onsavescript.sh cam1 $FILENAME
```
Works for me, and is pretty much equivalent to my old setup.

alexw22, 2021-02-02:
> My workaround has been a cronjob `*/5 * * * *` to make the snapshot; something like this:
>
> ```
> #!/bin/bash
> FILENAME="/home/_webcam/cam1/"$(date +'%Y-%m-%d')/$(date +'%Y%m%d-%H%M-%S--snapshot-0').jpg [...]

Thanks a lot for sharing! Looks like this grabs (with a timeout of 10 seconds) the current view every 5 minutes. Where does it save these? In the same folders as the motion pictures? Optimally i would like to keep the individual frames separate from the motion pictures and generate a video file at the end of each day that can be found in the video tab. Why did you have to add the entry to your motion.conf file?

degerrit, 2021-02-02:
@alexw22 `--output $FILENAME` sets the output folder and filename, e.g. you can save it where you want.  I use the `onsavescript.sh` for symlinking the latest pic for a separate mail routine I have, and some other stuff.
It's simple bash: just adapt it to your needs!