motioneye-project/motioneye #38: ./configure

piecam, 2015-11-06:
HI Chrisan,

Tried with git mrdave branch of motion and the following command; 
cd [your-motion-dir]
./configure --prefix=/usr --without-pgsql --without-sdl --without-sqlite3 --without-mysql

However, it said no such file /directory.
peng@cac:~/motion$ ./configure --prefix=/usr --without-pgsql --without-sdl --without-sqlite3 --without-mysql
-bash: ./configure: No such file or directory

Any thoughts on this?

I was trying to use a small form factor PC to be a central server to manage all remote pie cam and do a central storage and capture.

Please let me know.

thanks in advance
peng

ccrisan, 2015-11-06:
`motion-mrdave`'s configure mechanism has recently been changed. You'll need to run `autoreconf` in the source folder first and then run the normal `./configure...` command.

piecam, 2015-11-07:
HI Crisan,
Tried "autoconf" within that directory and did "./configure --prefix=/usr --without-pgsql --without-sdl --without-sqlite3 --without-mysql". Bumped into this error.

Any suggestions? Sorry to bother for this.

checking for libjpeg-mmx... skipping
checking for jpeg_set_defaults in -ljpeg... yes
./configure: line 5283: syntax error near unexpected token `FFMPEG,'
./configure: line 5283: `	PKG_CHECK_MODULES(FFMPEG, libavutil libavformat libavcodec libswscale, HAVE_FFMPEG=yes)'
root@cac:/home/peng/motion-mrdave# cat config
config.log    configure     configure.ac  
root@cac:/home/peng/motion-mrdave# cat configure | grep FFMPEG
FFMPEG_OBJ
  --with-ffmpeg=DIR       Build with FFMPEG support
FFMPEG_OBJ=""
	PKG_CHECK_MODULES(FFMPEG, libavutil libavformat libavcodec libswscale, HAVE_FFMPEG=yes)
if test "${HAVE_FFMPEG}" = "yes" ; then :
       FFMPEG_OBJ="ffmpeg.o"
$as_echo "#define HAVE_FFMPEG 1" >>confdefs.h
if test "${HAVE_FFMPEG}" = "yes"; then
    echo " ... FFMPEG_CFLAGS: $FFMPEG_CFLAGS"
    echo " ... FFMPEG_LIBS: $FFMPEG_LIBS"

ccrisan, 2015-11-07:
Yes, use autoreconf, as I suggested in my previous answer.

ccrisan, 2015-11-07:
The wiki page now mentions this autoreconf command.

closed 2015-11-07 by ccrisan

piecam, 2015-11-07:
thank you very much Crisan!

piecam, 2015-11-07:
oh Crisan,

What's the method to confirm that newly compiled "motion" is indeed used?

thanks
peng

ccrisan, 2015-11-07:
`motion -h` should report something else than 3.2.12.

piecam, 2015-11-08:
much appreciated!

christf, 2016-07-06:
I am running into the same error as piecam and used autoreconf to create a configure script:

./configure: line 5283: syntax error near unexpected token `FFMPEG,'
./configure: line 5283: `	PKG_CHECK_MODULES(FFMPEG, libavutil libavformat libavcodec libswscale, HAVE_FFMPEG=yes)'

the configure script reads at those lines:

if test "x$with_ffmpeg" != "xno"; then :

       if test "x$with_ffmpeg" != "xyes"; then :

              PKG_CONFIG_PATH=${with_ffmpeg}/lib/pkgconfig:$PKG_CONFIG_PATH
              export PKG_CONFIG_PATH

fi
        PKG_CHECK_MODULES(FFMPEG, libavutil libavformat libavcodec libswscale, HAVE_FFMPEG=yes)
fi

what can I do to fix this?

bakroistvan, 2016-07-29:
+1

I used `autoreconf` for mrdave edition and get the same error.

ccrisan, 2016-07-29:
@bakroistvan which one, more precisely?

bakroistvan, 2016-07-29:
I did the following after `sudo su` on CHIP (ARM v7):
```
apt-get install build-essential autoconf libjpeg-dev libavformat-dev libavcodec-dev libswscale-dev ffmpeg subversion git
git clone https://github.com/Mr-Dave/motion.git motion-mrdave

cd motion-mrdave
autoreconf
./configure --prefix=/usr --without-pgsql --without-sdl --without-sqlite3 --without-mysql
```

```
$ ffmpeg
ffmpeg version 3.0.2-4~bpo8+1 Copyright (c) 2000-2016 the FFmpeg developers
  built with gcc 4.9.2 (Debian 4.9.2-10)
  configuration: --prefix=/usr --extra-version='4~bpo8+1' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librubberband --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-chromaprint --enable-libx264
  libavutil      55. 17.103 / 55. 17.103
  libavcodec     57. 24.102 / 57. 24.102
  libavformat    57. 25.100 / 57. 25.100
  libavdevice    57.  0.101 / 57.  0.101
  libavfilter     6. 31.100 /  6. 31.100
  libavresample   3.  0.  0 /  3.  0.  0
  libswscale      4.  0.100 /  4.  0.100
  libswresample   2.  0.101 /  2.  0.101
  libpostproc    54.  0.100 / 54.  0.100
Hyper fast Audio and Video encoder
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...

Use -h to get full help or, even better, run 'man ffmpeg'

```

I get the same error as @christf had. I tried with the SVN, it works.
So I want to mention that for me: Mr Dave's fork is not recommended.

DeanCording, 2016-08-02:
apt-get install pkgconf

Fixes the problem.

bakroistvan, 2016-08-03:
Yes it does. Thank you