motioneye-project/motioneye #39: web video flash/jump fast

piecam, 2015-11-07:
HI Crisan,

I don't know how to describe this. I had Pi/Cam as networkcam/simple mjpeg cam connected to my new Small PC with motioneeye installed and take in streaming with URL of "IP:8081". I have another pi cam with motionpie installed and remotely connected to this new pc as well as a "motioneye" peer.

The image I have on my Web view of those streaming are "jumping" or "refreshing" fast. It's not a smooth display.

Is there any parameters that I didn't put it correct value?

thanks in advance.
peng

piecam, 2015-11-07:
HI Crisan,

Here's a quick screen video capture. Any thoughts about what I have to adjust?

https://up.f2f10.com/index.php/s/Cws8o60er8tubCI

Thanks a lot
Peng

ccrisan, 2015-11-09:
The behaviour that I see in your screencast is not normal. To be honest I never saw something like that with motionEye. Does it happen with other browsers as well?

pghpete, 2015-11-09:
I'm having the same problem with Firefox on a linux box but Chromium doesn't have the same issue in my situation. Also I only have 1 camera and the 1 image isn't being resized in firefox like it is nicely on chromium.  Initially I thought this might have to do with the "power line frequency" set in the motion.conf file but that no longer makes sense. (and didn't make a difference in my situation. I'm wondering if it's something flash related within firefox.

pghpete, 2015-11-09:
Peng, what OS are you running? (and kernel) Firefox version? Flash plugin version etc? I was trying to track down this issue and gave up but shortly afterwards updated to newest firefox and kernel and i'm not having the problem anymore. (of flashing camera feeds) (the resize of the camera blocks though still isn't working correctly for me.... but the flashing is gone and that was my major issue.

piecam, 2015-11-09:
HI Guys,
I was on linux/mint/elementary os.
Just upgraded FF to ver 42 and it seems now ok.

Thanks for helping out.

Peng

closed 2016-07-03 by ccrisan