motioneye-project/motioneye #40: Add pagination to movies/pictures

chris1705, 2015-11-07:
Hello ccrisan,

would it be possible to add pagination (or maybe continuous scrolling?) to the pictures and movies? At some days my cameras take more than 10.000 pictures and it takes a long time to load them. Sometimes the list even stops loading. (The error I receive while loading the list from /picture/{id}/list/ is: `{"error": "Failed to get pictures list."}`)

Regards
Chris

maxholgasson, 2016-05-17:
is there any news about it?

MobilisInMobilum, 2019-08-12:
+1, very slow load when opening gallery with few hundreds of pictures, adding pagination would solve the issue. Also listing images in carousel sometimes times out with no apparent reason, its like specific image request hangs out, in such case I cannot even refresh the page, it still hans out, and I have to close the tab and open it again.

nicolasluckie, 2019-10-16:
+1, I take 86,400 photos per day (1 per second), as opposed to continuous recording. By the end of the day, I can't preview any of the photos because any browser I use grinds to a halt.

JBS5, 2020-01-07:
+1. Any news on this feature request?