motioneye-project/motioneye #41: Ability to use raspistill while motioneye is running?

pghpete, 2015-11-07:
I am not sure this is an issue with motioneye, raspistill, or by design but I thought I would ask here.  Is it possible (or should it be possible) to run the raspberry pi raspistill camera utility when motioneye is running?  Motion/Motioneye uses /dev/video0 and I thought I was able to run the raspistill to capture still images with motioneye running in the past but I can no longer do that. [Raspberry Pi 2 Model B, Raspian Jesse, Pi NOIR camera module]

With motioneye off, "raspistill -o image.jpg" works as expected, with motioneye on, raspistill errors out with the following...

pi@seccam ~ $ raspistill -o image.jpg
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

I have updated Raspian and the Pi's firmware to the latest available software/firmware.

ccrisan, 2015-11-07:
I doubt that would be possible, as the mmal interface is already in use by the v4l2 module.

pghpete, 2015-11-07:
What a fast response lol thank you!  Perhaps a feature request to alow you to take still images from the gui is in order? :)  On a side note, is there any way I can contribute back to this project? Be it something small like a man page or documentation, or debugging, testing, etc?

ccrisan, 2015-11-07:
Yes, the feature has already been proposed (I'm not sure if there's an actual issue about it, but I remember users asking for taking snapshots from the UI).

Well since you mentioned the key word "documentation" (:P), motionEye could really use some reference material when it comes to the settings panel. If you could take each available setting from the settings panel, per section, and document it (possibly linking to the corresponding [motion config option](http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigFileOptions), it would be great.

As soon as I have some content regarding this matter, I could turn it into a settings wiki page.

pghpete, 2015-11-07:
Sound good. I'll spend some time and see what I can provide you.  Where should I send the resulting info and in what format?

ccrisan, 2015-11-07:
You can send it directly to me (ccrisan around gmail) in whatever format you want. I will convert it to markdown then.

pghpete, 2015-11-08:
fyi - sent some items via email. holler if I can improve them or update them for you.

chrisf4lc0n, 2015-12-04:
Hi, let me chip in.
I am running a remote doorbell with my Pi 1 and planning to run MotionEye alongside. I have just thought of a workaround... I can script, that when the button on the bell gets pressed it shuts the MotionEye down, rings the bell, takes photo and restart the MotionEye.
Maybe that can help... Once I have MotionEye up and running I will report back with more details.

chrisf4lc0n, 2015-12-06:
I had a similar problem, as I was running my Doorbell script which was taking a static photo when Doorbell was pressed, but when stream was running I could not use the camera. I sorted it out writing a simple script, for you could be a bit different but that should help:

sudo nano raspistill.sh

add:
#!/bin/sh
d=`date +%d-%m-%y`
t=`date +%T`
systemctl stop motioneye.service
raspistill -o $t$d.jpg -w 1024 -h 768 -q 30 -hf
sleep 15
systemctl start motioneye.service
exit

sudo chmod a+x raspistill.sh

That will shut the motioneye service for 15 seconds, takes photo and restarts the service.
You can obviously modify it to your needs.
If you need something more advanced let me know.

chrisf4lc0n, 2015-12-20:
There is another way of achieving that by either using https://github.com/umlaeute/v4l2loopback or https://github.com/mpromonet/v4l2tools unfortunately the 1st one does not seem to be working with any 4.x kernel and the 2nd one does not create another device with my camera module. You can however test it and see if it works, that way you will have 2 separate camera streams with 1 camera allowing you to take photos and record/stream at the same time.