motioneye-project/motioneye #42: motioneye continuously fill journald

jeanmarc77, 2015-11-08:
Hello,
motioneye continuously fill journald, it is possible to make it quiet ?
Thanks !

[root@rpi2 ~]# journalctl -f
-- Logs begin at Wed 2015-09-02 21:08:23 CEST. --
Nov 08 09:02:04 rpi2 motion[30269]: [1] netcam_proc_jpeg: processing jpeg image - content length 21380
Nov 08 09:02:04 rpi2 motion[30269]: [1] netcam_proc_jpeg: processing jpeg image - content length 22768
Nov 08 09:02:04 rpi2 motion[30269]: [1] netcam_proc_jpeg: processing jpeg image - content length 22804
Nov 08 09:02:04 rpi2 motion[30269]: [1] netcam_proc_jpeg: processing jpeg image - content length 21380
Nov 08 09:02:04 rpi2 motion[30269]: [1] netcam_proc_jpeg: processing jpeg image - content length 21288
Nov 08 09:02:04 rpi2 motion[30269]: [1] netcam_proc_jpeg: processing jpeg image - content length 21308
.. and so on

ccrisan, 2015-11-08:
Have you enabled debugging? What does your motioneye.conf file look like? How do you start motionEye?

jeanmarc77, 2015-11-08:
Hi,
Thanks for your fast reply. I haven't enable debugging, i start motioneye with :  systemctl start motioneye.
my motioneye.conf is
# @enabled on
# @show_advanced on
# @normal_password 
# @admin_username admin
# @admin_password xxx
# @normal_username user

control_html_output on
control_localhost on
control_port 7999

thread thread-1.conf

Tell me if i can test anything to help.

ccrisan, 2015-11-08:
What version of motionEye do you run?

jeanmarc77, 2015-11-08:
0.28.3, i did a pip2 install motioneye --upgrade

ccrisan, 2015-11-08:
What does `ps aux | grep motion` say?

jeanmarc77, 2015-11-09:
Hi,

[root@rpi2 ~]# ps aux | grep motion
root     22492 14.5  1.8  24244 17396 ?        Ss   07:38   0:01 /usr/bin/python2 /usr/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
root     22546 78.4  1.6  69096 15352 ?        Sl   07:38   0:07 /usr/sbin/motion -n -c /etc/motioneye/motion.conf -d 6
root     22924  0.0  0.0   2652   592 pts/0    S+   07:38   0:00 grep motion

ccrisan, 2015-11-09:
Well the command looks alright. The odd thing is that `-d 6` is the default and should not fill up the log. At least mine doesn't. What version of motion do you use, by the way?

jeanmarc77, 2015-11-09:
It's 3.2.12 version. For now, i've put my journal as volatile to avoid ruin my sd card ;)

ccrisan, 2015-11-09:
Could you please use Mr Dave's version of motion or the SVN version?

jeanmarc77, 2015-11-09:
I've been unable to compile properly the svn version

[root@rpi2 3.2]# make
Welcome to the setup procedure for Motion, the motion detection daemon! If you get
error messages during this procedure, please report them to the mailing list. The
Motion Guide contains all information you should need to get Motion up and running.
Run "make updateguide" to download the latest version of the Motion Guide.

Version: 3.2.12
Platform: Linux (if this is incorrect, please read README.FreeBSD)

Generating dependencies, please wait...
In file included from motion.c:10:0:
motion.h:43:28: fatal error: linux/videodev.h: No such file or directory

So i used the Mr dave git. I currently use Unofficial-Git-2caced3, which is indeed way less verbose.

ccrisan, 2015-11-09:
I have updated the default log level to 5 so that older motion binaries won't fill up the log.

closed 2015-11-09 by ccrisan