motioneye-project/motioneye #43: Still images are saved but not movies during motion detection

garyni, 2015-11-09:
Hello,

Can anyone help me with this? No movies create on motion detection. Build is based on this article https://bitbucket.org/ccrisan/motioneye/wiki/Installation%20Guide .Log file as follows

Many Thanks
GAry

[2] netcam_next called with no data in buffer
[1] netcam_next called with no data in buffer
[2] netcam_next called with no data in buffer
[2] netcam_next called with no data in buffer
[2] netcam_next called with no data in buffer
[2] netcam_next called with no data in buffer
[2] netcam_next called with no data in buffer
[2] netcam_next called with no data in buffer
[2] netcam_next called with no data in buffer
[2] ffmpeg_avcodec_log: Value 4707126720094797824.000000 for parameter 'probesize' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4707126720094797824.000000 for parameter 'analyzeduration' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4697254411347427328.000000 for parameter 'indexmem' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4703785510416416768.000000 for parameter 'rtbufsize' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'fpsprobesize' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4607182418800017408.000000 for parameter 'f_err_detect' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4607182418800017408.000000 for parameter 'err_detect' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4686111960511545344.000000 for parameter 'b' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4683532506232782848.000000 for parameter 'ab' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4705844345939427328.000000 for parameter 'bt' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4617315517961601024.000000 for parameter 'me_method' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4622945017495814144.000000 for parameter 'g' out of range
[2] ffmpeg_avcodec_log: Value 4611686018427387904.000000 for parameter 'qmin' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4629418941960159232.000000 for parameter 'qmax' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4613937818241073152.000000 for parameter 'qdiff' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'wpredp' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4607182418800017408.000000 for parameter 'bug' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4607182418800017408.000000 for parameter 'er' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4607182418800017408.000000 for parameter 'err_detect' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4613937818241073152.000000 for parameter 'ec' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4620693217682128896.000000 for parameter 'ildctcmp' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4620693217682128896.000000 for parameter 'subq' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4696837138094751744.000000 for parameter 'ibias' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4696837138094751744.000000 for parameter 'pbias' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4642507528377204736.000000 for parameter 'lmin' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4660262442142531584.000000 for parameter 'lmax' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4696130710463840256.000000 for parameter 'flags2' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4607182418800017408.000000 for parameter 'threads' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4620693217682128896.000000 for parameter 'nssew' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4586705114244317184.000000 for parameter 'profile' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4586705114244317184.000000 for parameter 'level' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4623507967449235456.000000 for parameter 'skipcmp' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4642507528377204736.000000 for parameter 'mblmin' out of range
- flag 16
[2] ffmpeg_avcodec_log: Value 4660262442142531584.000000 for parameter 'mblmax' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4643211215818981376.000000 for parameter 'mepc' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4607182418800017408.000000 for parameter 'bidir_refine' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'cqp' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4627730092099895296.000000 for parameter 'keyint_min' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4607182418800017408.000000 for parameter 'refs' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'directpred' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4618441417868443648.000000 for parameter 'sc_factor' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4643211215818981376.000000 for parameter 'mv0_threshold' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4630826316843712512.000000 for parameter 'b_sensitivity' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'compression_level' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'min_prediction_order' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'max_prediction_order' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'prediction_order_method' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'min_partition_order' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'max_partition_order' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4607182418800017408.000000 for parameter 'ticks_per_frame' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4611686018427387904.000000 for parameter 'color_primaries' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4611686018427387904.000000 for parameter 'color_trc' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4611686018427387904.000000 for parameter 'colorspace' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'aq_mode' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'rc_lookahead' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'lpc_type' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'lpc_passes' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value 4613937818241073152.000000 for parameter 'thread_type' out of range
 - flag 16
[2] ffmpeg_avcodec_log: Value -4616189618054758400.000000 for parameter 'request_sample_fmt' out of range
 - flag 16
[2] ffmpeg_avcodec_log: err{or,}_recognition separate: 0; 0
 - flag 48
[2] ffmpeg_avcodec_log: err{or,}_recognition combined: 0; 0
 - flag 48
[2] ffmpeg_avcodec_log: bitrate tolerance too small for bitrate
 - flag 16
[2] avcodec_open - could not open codec: Operation now in progress
[2] ffopen_open error creating (new) file [/data/media/cam1/2015-11-09/12-06-53.avi]: Operation now in progress
[2] File of type 1 saved to: /data/media/cam1/2015-11-09/12-06-52.jpg
[2] File of type 1 saved to: /data/media/cam1/2015-11-09/12-06-52.jpg
[2] Thread exiting
[2] Calling vid_close() from motion_cleanup
[2] vid_close: calling netcam_cleanup
[2] netcam camera handler: finish set, exiting

ccrisan, 2015-11-09:
You should follow [this](https://github.com/ccrisan/motioneye/wiki/Installation) guide if working with newer versions of motionEye (>= 0.27). What versions of ffmpeg, motion and motionEye do you use?

garyni, 2015-11-09:
Hi, thanks for the response

This is an install on Raspbian.

root@raspberrypi:~# ffmpeg -version
ffmpeg version 1.0.10
built on Jul 25 2014 08:30:08 with gcc 4.6 (Debian 4.6.3-14)

motion Version 3.2.12

motionEye 0.26.

garyni, 2015-11-11:
I've re-installed with the latest version and this is resolved. Thanks

closed 2015-11-11 by garyni