motioneye-project/motioneye #44: Acces lost

jerome33, 2015-11-11:
Hello, 

I installed Motioneye on Raspbian with an acces point config (dhcpd and isc-dhcp-serve), so I can directly connect MotionEye by WIFI. 
Everything goes OK, until I reboot the Raspberry... then I lost http access : ERR_CONNECTION_REFUSED ... but I still can acces the Pi by SSH...

Motioneye service is running ok, dhcpd running ok and isc-dhcp-server is running ok too...

Do you have an idea ? 

Thanks a lot for this good job !

ccrisan, 2015-11-11:
How do you know motionEye is running? What does `ps aux | grep motioneye` say?

jerome33, 2015-11-11:
pi         879  0.0  0.1   4596  1740 pts/0    S+   19:00   0:00 grep --color=auto motioneye     (<- in red)

jerome33, 2015-11-11:
service motioneye status

* motioneye.service - LSB: Start the motionEye server
   Loaded: loaded (/etc/init.d/motioneye)
   Active: active (exited) since mar. 2015-11-10 18:58:09 CET; 4min 37s ago
  Process: 390 ExecStart=/etc/init.d/motioneye start (code=exited, status=0/SUCCESS)

ccrisan, 2015-11-11:
How about `journalctl -u motioneye`?

pghpete, 2015-11-11:
That grep line is only the ps aux | grep reseults you ran.

try
systemctl enable motioneye (will make sure it starts at boot)
systemctl start motioneye
then check with
systemctl enable motioneye

Also check your logs in /var/log/motioneye.log if you have set it up to write there.

ccrisan, 2015-11-11:
That's the point of `journalctl -u motioneye`, as I believe Jerome didn't configure his setup to log to /var/log.

jerome33, 2015-11-11:
1/ journalctl -u motioneye :
-- Logs begin at mar. 2015-11-10 19:09:23 CET, end at mar. 2015-11-10 19:21:24 CET. --
nov. 10 19:09:25 simon-computer systemd[1]: Starting LSB: Start the motionEye server...
nov. 10 19:09:30 simon-computer motioneye[394]: Starting motionEye server:.
nov. 10 19:09:30 simon-computer systemd[1]: Started LSB: Start the motionEye server.

2/ systemctl enable motioneye ...

pi@simon-computer ~ $ sudo systemctl enable motioneye
Synchronizing state for motioneye.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d motioneye defaults
Executing /usr/sbin/update-rc.d motioneye enable
pi@simon-computer ~ $ sudo systemctl start motioneye
pi@simon-computer ~ $ sudo systemctl enable motioneye
Synchronizing state for motioneye.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d motioneye defaults
Executing /usr/sbin/update-rc.d motioneye enable

No http access...

3/ the log file : 

2015-11-10 14:04:14: [motioneye]     INFO: hello! this is motionEye server 0.28.3
2015-11-10 14:04:15: [motioneye]     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
2015-11-10 14:04:15: [motioneye]     INFO: cleanup started
2015-11-10 14:04:15: [motioneye]     INFO: wsswitch started
2015-11-10 14:04:15: [motioneye]     INFO: thumbnailer started
2015-11-10 14:04:15: [motioneye]     INFO: mjpg client garbage collector started
2015-11-10 14:04:15: [motioneye]     INFO: server started
2015-11-10 18:43:21: [motioneye]     INFO: interrupt signal received, shutting down...
2015-11-10 18:43:21: [motioneye]     INFO: server stopped
2015-11-10 18:43:21: [motioneye]     INFO: bye!
2015-11-10 18:43:34: [motioneye]     INFO: hello! this is motionEye server 0.28.3
2015-11-10 18:43:35: [motioneye]     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
2015-11-10 18:43:35: [motioneye]     INFO: cleanup started
2015-11-10 18:43:35: [motioneye]     INFO: wsswitch started
2015-11-10 18:43:35: [motioneye]     INFO: thumbnailer started
2015-11-10 18:43:35: [motioneye]     INFO: mjpg client garbage collector started
2015-11-10 18:43:35: [motioneye]     INFO: server started
2015-11-10 18:45:41: [motioneye]     INFO: adding new camera with id 1...
2015-11-10 18:45:41: [motioneye]     INFO: main config file /etc/motioneye/motion.conf does not exist, using default values
2015-11-10 18:53:16: [motioneye]     INFO: interrupt signal received, shutting down...
2015-11-10 18:53:16: [motioneye]     INFO: server stopped
2015-11-10 18:53:18: [motioneye]     INFO: motion stopped
2015-11-10 18:53:18: [motioneye]     INFO: bye!
2015-11-10 18:53:18: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
2015-11-10 18:53:23: [motioneye]     INFO: hello! this is motionEye server 0.28.3
2015-11-10 18:58:09: [motioneye]     INFO: hello! this is motionEye server 0.28.3
2015-11-10 19:09:29: [motioneye]     INFO: hello! this is motionEye server 0.28.3

ccrisan, 2015-11-11:
Please run the command manually: `sudo meyectl startserver -c /etc/motioneye/motioneye.conf -d` and see if it works.

jerome33, 2015-11-11:
Give me this (and don't work :( )

    INFO: hello! this is motionEye server 0.28.3
   DEBUG: reading main config from file /etc/motioneye/motion.conf...
   DEBUG: loading additional config structure for main, without separators
   DEBUG: additional config section: network
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 9, in <module>
    load_entry_point('motioneye==0.28.3', 'console_scripts', 'meyectl')()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/meyectl.py", line 248, in main
    server.main(arg_parser, sys.argv[2:], command[:-6])
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 409, in main
    run()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 340, in run
    start_motion()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 308, in start_motion
    motionctl.start()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 84, in start
    enabled_local_motion_cameras = config.get_enabled_local_motion_cameras()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 218, in get_enabled_local_motion_cameras
    if not get_main().get('@enabled'):
  File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 120, in get_main
    _get_additional_config(main_config)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 1726, in _get_additional_config
    (sections, configs) = get_additional_structure(camera=bool(camera_id))
  File "/usr/local/lib/python2.7/dist-packages/motioneye/config.py", line 1697, in get_additional_structure
    result = func()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/tzctl.py", line 126, in timeZone
    import pytz
ImportError: No module named pytz

ccrisan, 2015-11-11:
Have you enabled timezone control in your motioneye.conf?

jerome33, 2015-11-11:
yes... My raspberry will be used without internet and I'd like to have the time setting enabled...

ccrisan, 2015-11-11:
In that case, `pip install pytz`.

jerome33, 2015-11-11:
it works ! I tried to reboot several times and all seems to be ok :dancer: 

Thanks a lot !!!

closed 2016-07-03 by ccrisan