motioneye-project/motioneye #45: Q: File Permissions

luetzel, 2015-11-11:
Hi Calin,
I have a small question regarding file permissions of the media_path. I'm running motioneye on Raspian as user 'root', so that all files in the media_path are stored under /home/pi/pictures with 'root' ownership. Although all jpg files have -rw-rw-rw- permissions, I'm unable to delete them as regular user 'pi' on the console, because permissions of the parent directories '2015-11-11'  are set to drwxr-xr-x.

Is it possible to configure file permissions for the media folder somewhere? I have mounted my Raspberry's pictures folder as a samba share. When I try to delete pictures, the Window Explorer also complains about insufficient permissions. 

I know, that it is possible to run motion eye as the user 'pi', but this is not an option for me, because it disables the possibility to 'reboot' through the web interface.

Cheers
/luetzel

ccrisan, 2015-11-11:
You'll need to play with `umask` in order to change the created files permissions. However I often find it difficult to deal with umask settings, especially when working with processes and subprocesses (which is the case of motionEye).

If you must have full access to your media files using your regular user, I suggest [running motionEye as a normal user](https://github.com/ccrisan/motioneye/wiki/Running-As-Normal-User).

luetzel, 2015-11-11:
Thank you for the fast reply.
Can you point my to place where to set umask for motioneye? I found os.umask(0) in server.py, but I'm not sure whether it is the right place to configure it ...

ccrisan, 2015-11-12:
It's not a python thing, but rather la Linux thing. I honestly don't know where to change that setting on a per process basis (I would edit the `/etc/login.defs` file). However I strongly recommend that you run motionEye as a normal user, in your case.

luetzel, 2015-11-12:
Ok, thanks. I tried to set umask at different places without success. I couldn't find settings that affect subprocesses as well. I'll try to find a workaround that chmods the directories within my pictures folder.
I cannot run it as a normal user, because I need the "shutdown/reboot" function of the web interface.

closed 2015-11-12 by luetzel