motioneye-project/motioneye #46: Firefox on Linux: Picutre/Movies dialog

pghpete, 2015-11-11:
When I use Firefox (v42) on linux and I click a camera feeds icon for pictures or moves it shows up like this...

http://i.imgur.com/V4C7VQn.png

I cannot exit (no exit button), no folders on the left side, and no buttons on the bottom.  I have to reload the page and sign in again.

( a side issue, not as important is that in Chromium it resizes the camera feed block very nicely but that doesn't seam to be happening in Firefox)

Thoughts?

luetzel, 2015-11-12:
I can confirm this issue with version 0.28.3. It seems that only Firefox v42 is affected. Internet Explorer works fine as well.

closed 2015-11-13 by ccrisan

pghpete, 2015-11-13:
I checked for an upgrade to motioneye via pip an didn't see one.  Do we know where/what is causing this and what a possible solution may be for this problem? Not sure why it's closed now as the problem still exists for me.

ccrisan, 2015-11-13:
Most likely there is no new version. This fix hasn't been included in a release yet.