motioneye-project/motioneye #47: Mask Question

stuartbloom, 2015-11-12:


stuartbloom, 2015-11-12:
I know you are working on masks, but I have just created a pgm file for a mask, copied it to /etc/motioneye and added the following line to the thread-1.cong file:

mask_file thread-1_mask.pgm

Whenever I apply any new settings, this value gets lost.  Is there any other way of doing this or will I have to wait for the implementation of masks?

Thanks in advance

Stuart

ccrisan, 2015-11-12:
Use the extra motion settings field. It should work (although I haven't tested this case).

stuartbloom, 2015-11-12:
thanks appreciate that. will have a play :)

closed 2015-11-12 by stuartbloom

ozett, 2015-12-25:
did i worked ok? can i go this way, too?

stuartbloom, 2016-01-08:
sorry for the delay.yes, it did work initially, but for some reason it is no longer working.  I need to do some more investigation to see what's wrong. But i suspect its something in motion rather than motioneye