motioneye-project/motioneye #52: problem with snapshot

mferlitsch, 2015-11-18:
Hello,

I tried to add a cam (1920x1080) to motioneye.

But I always get this error:

Nov 18 09:20:19 raspberrypi motion: [2] Started stream webcam server in port 8082
Nov 18 09:20:19 raspberrypi motion: [2] Resizing pre_capture buffer to 3 items
Nov 18 09:20:20 raspberrypi motion: [2] Retrying until successful connection with camera
Nov 18 09:20:20 raspberrypi motion: [2] Removed netcam Keep-Alive flagdue to apparent closed HTTP connection.
Nov 18 09:20:25 raspberrypi motion: [2] libjpeg decompression failure on first frame - giving up!
Nov 18 09:20:30 raspberrypi motion: [2] Retrying until successful connection with camera
Nov 18 09:20:30 raspberrypi motion: [2] Removed netcam Keep-Alive flagdue to apparent closed HTTP connection.
Nov 18 09:20:35 raspberrypi motion: [2] libjpeg decompression failure on first frame - giving up!
Nov 18 09:20:40 raspberrypi motion: [2] Retrying until successful connection with camera
Nov 18 09:20:40 raspberrypi motion: [2] Removed netcam Keep-Alive flagdue to apparent closed HTTP connection.
Nov 18 09:20:45 raspberrypi motion: [2] libjpeg decompression failure on first frame - giving up!

How can I solve this issue?

ccrisan, 2015-11-18:
motion doesn't work with resolutions that are not modulo 16 (1080 / 16.0 = 67.5). What camera do you use and how is it connected to the motionEye server (USB, netcam?)

ccrisan, 2015-11-19:
I don't know about other motion-like tools. RTSP is supported but has some issues with the latest stable version. Just wait fot 0.28.4 which will fix RTSP support. You'll need a recent motion version though.

What do you mean 0.27 does not start? How did you install and start it? Did you follow the [instructions](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian)?.

mferlitsch, 2015-11-19:
Yes I followed the instructions. The 0.24 runs. When I start the 0.27 I always get the path and the program seems to be closed - there is no output on console.
Then I tried the 0.26 - it started but there the RTSP does not work fine.

Has the 0.28.4 the same problem with higher resolutions or can it work with every resolution?

ccrisan, 2015-11-19:
The 0.27 version has been radically redesigned. Nevertheless it must work.

If you followed the [instructions](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian) I gave you, then you couldn't install 0.24, as they are available from 0.27 on and will install, by default the 0.28.3 version.

If you say you have tried motionEye 0.27, how did you start it?

mferlitsch, 2015-11-19:
I started it via console with ./motioneye.py
Then I got back the console - when I started the 0.24 I did not get back the input-path - instead I got some messages of detected cams and so on - with STRG-C I closed the app.

ccrisan, 2015-11-19:
I'm sorry but I don't think you have followed the [instructions](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian). motionEye is no longer started using ./motioneye.py.

closed 2015-11-19 by ccrisan