motioneye-project/motioneye #53: Date / Time field

jerome33, 2015-11-23:
Hello, 

Me again...

Do you have the possibility to add a date/time field to the advanced settings, to send a "sudo date" command to the system? My system is stand alone, with no Internet, and at each start, I must use  ssh to set the date. It will be vey usefull (for me, and maybe others ;) ) 

Regards

Jérome

sgiannulli, 2015-11-23:
Hello Jerome,
believe me the best solution is you add an RTC module in your raspberry
it is very cheap http://www.ebay.it/itm/Ds3231-Reloj-tiempo-real-3-3-v-5v-Alta-Precision-Para-Raspberry-Pi-Clock-/331663047691?hash=item4d38a87c0b:g:CMEAAOSwuTxWAuBy
and realy easy to install https://www.raspberrypi.org/forums/viewtopic.php?p=775626#p775626
this will avoid many problems on server side

ccrisan, 2015-11-23:
@jerome33, it wouldn't help too much. As @sgiannulli said, in the absence of an internet connection and without a dedicated RTC module, the drift quickly becomes so big that it doesn't make any sense to consider the local time at all.

closed 2015-11-23 by ccrisan