motioneye-project/motioneye #54: Variables for notifications

mneumark, 2015-11-25:
It would be really nice if you could pass variables to notifications, like say, a URL that points to an image taken with the notification, and a URL that points to the recorded video. Perhaps notification time, camera name. Things like that. Stuff you could access from the notification command and the Web Hook Notification.

ccrisan, 2015-11-25:
The problem is that the `%f` [conversion specifier] (http://www.lavrsen.dk/foswiki/bin/view/Motion/ConversionSpecifiers) is not available with the `on_event_end` directive, which is used for notification. I'll think of a workaround.

P.S. upload to Google Drive and Dropbox are on the way and I'm thinking of adding generic HTTP upload support as well. I believe that's what you're actually looking for.

mneumark, 2015-11-25:
Actually I was trying to create a push notification system using PushOver. I wanted the push notification in PushOver to contain an image. PuahOver supports a message with HTML so all I need is an image URL and once you do that you may as well add in a couple other variables. Something like:
```bash
curl api.blah.com/push?message=Motion%20detected%20in%20{{camera_name}}&image={{image_url}}
```

ccrisan, 2015-11-27:
I just added support for webhook calling and command running for each media file created. The feature will be included in the next 0.29 release.

closed 2015-11-27 by ccrisan

mneumark, 2015-11-27:
Sweet, that should help.

MattMulhern, 2018-06-26:
_Sorry for commenting on a closed issue, but there doesn't seem to be a similar open one._

I understand that the "run a command" and "run a webhook"have been added to the media upload section which is awesome. 

Are there any default xargs or environment vars set in the shell running your command specified in the "run a command" dialog? I've got a simple bash script im running and was wondering if there was a handy way to know the actual filepath (just on local disk, not where it's being uploaded to) of the media file being uploaded.

ccrisan, 2018-06-26:
You can use the %f conversion specifier. It's documented there if you hover the question mark next to the field.