motioneye-project/motioneye #55: avi.thumb files will not be deleted

hthiery, 2015-11-25:
when deleting a movie file the corresponding thumb file is still in the output folder

ccrisan, 2015-11-25:
What version of motionEye are you using? What is (was) the path of the movie?

hthiery, 2015-11-25:
meye-caa8f060 from motioneyeos for rasperry pi

hthiery, 2015-11-25:
can i find a version string in webinterface?

hthiery, 2015-11-25:
Current Version: 20151103

ccrisan, 2015-12-03:
You didn't answer my question: what was the path of the movie? What is the full path of the .thumb file? Are there any other files in that folder?

hthiery, 2015-12-03:
Currently I don't have access to the system. But as far as I know it is the default path. I added two cameras. Both write their files (avi and jpg) to that location.

ccrisan, 2015-12-03:
When you get back to the... system, could you please try to see if the issue is reproducible? As far as I can tell from my code, it should delete the thumb files as well. In case you manage to reproduce the issue, could you please enable the debug logging and send me the log file?

closed 2015-12-26 by ccrisan