motioneye-project/motioneye #57: MotionEye Fails to Start after new installation

bjs-pdx, 2015-11-25:
I just installed the latest version of MotionEye and get the following when I try to start it:

root@laptoptoo:~# /etc/init.d/motioneye start
 * Starting motionEye server                                                    Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 7, in <module>
    from motioneye.meyectl import main
  File "/usr/local/lib/python3.4/dist-packages/motioneye/meyectl.py", line 28, in <module>
    import settings
ImportError: No module named 'settings'
                                                                         [fail]
I've done some searching to find a solution without success.  What am I missing?

ccrisan, 2015-11-26:
motionEye does not work with python 3.4. Please make sure you follow the entire installation procedure with a python 2.7 installation.

closed 2015-11-26 by ccrisan

rainman-xx, 2017-01-07:
Hi, I'm using the latest motion eye using the instruction here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Also it looks like it is now using the python 3.x. But i'm still getting the same error:

`root@osmc:~#` /usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf
Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 11, in <module>
    load_entry_point('motioneye==0.35.1', 'console_scripts', 'meyectl')()
  File "/usr/local/lib/python3.4/dist-packages/pkg_resources/__init__.py", line 561, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python3.4/dist-packages/pkg_resources/__init__.py", line 2631, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python3.4/dist-packages/pkg_resources/__init__.py", line 2291, in load
    return self.resolve()
  File "/usr/local/lib/python3.4/dist-packages/pkg_resources/__init__.py", line 2297, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.4/dist-packages/motioneye/meyectl.py", line 28, in <module>
    import settings
ImportError: No module named 'settings'
`
Any advice?

ccrisan, 2017-01-07:
Use Python2.7. Python 3.x is not supported.

rainman-xx, 2017-01-07:
Thanks for the feedback.

However, why is the file `/usr/local/bin/meyectl` executed by python3 as indicated in the files first line `#!/usr/bin/python3` if it's not compatible with python 3.x

Now back to the problem.  I followed the instructions [here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian). Just to make sure I edited meyectl to run python 2.7 (2.7.9) changed `#!/usr/bin/python3` to `#!/usr/bin/python2`.

I started the service:
`sudo systemctl restart motioneye`

I get this result from the ` journalctl -x` (running in the separate console):

```
Jan 07 23:26:48 osmc systemd[1]: Starting motionEye Server...
-- Subject: Unit motioneye.service has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit motioneye.service has begun starting up.
Jan 07 23:26:48 osmc systemd[1]: Started motionEye Server.
-- Subject: Unit motioneye.service has finished start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit motioneye.service has finished starting up.
--
-- The start-up result is done.
Jan 07 23:26:48 osmc sudo[18006]: pam_unix(sudo:session): session closed for user root
Jan 07 23:26:48 osmc meyectl[18009]: Traceback (most recent call last):
Jan 07 23:26:48 osmc meyectl[18009]: File "/usr/local/bin/meyectl", line 6, in <module>
Jan 07 23:26:48 osmc meyectl[18009]: from pkg_resources import load_entry_point
Jan 07 23:26:48 osmc meyectl[18009]: File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2876, in <module>
Jan 07 23:26:48 osmc meyectl[18009]: working_set = WorkingSet._build_master()
Jan 07 23:26:48 osmc meyectl[18009]: File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 449, in _build_master
Jan 07 23:26:48 osmc meyectl[18009]: ws.require(__requires__)
Jan 07 23:26:48 osmc meyectl[18009]: File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 745, in require
Jan 07 23:26:48 osmc meyectl[18009]: needed = self.resolve(parse_requirements(requirements))
Jan 07 23:26:48 osmc meyectl[18009]: File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 639, in resolve
Jan 07 23:26:48 osmc meyectl[18009]: raise DistributionNotFound(req)
Jan 07 23:26:48 osmc meyectl[18009]: pkg_resources.DistributionNotFound: motioneye==0.35.1
Jan 07 23:26:48 osmc systemd[1]: motioneye.service: main process exited, code=exited, status=1/FAILURE
Jan 07 23:26:48 osmc systemd[1]: Unit motioneye.service entered failed state.

```

Where do you think the problem is? Thanks in advance.

ccrisan, 2017-01-07:
You have probably installed motionEye using python3's pip. On systems that default to python3, there's usually a pip2 command corresponding to python2. I'd recommend repeating the install procedure using pip2.

rainman-xx, 2017-01-09:
ccrisan,

It works. I think, the instruction has to be updated to specify pip2 instead of just pip since other system defaults to python3.

Raywes88, 2017-04-07:
I know it's my own fault for not double checking but can we update the command step 7 on this wiki page https://github.com/ccrisan/motioneye/wiki/Install-On-Arch to say pip2. Might save others a headache.

ccrisan, 2017-04-08:
@Raywes88 updated. Thanks!

jonathanbower, 2017-05-01:
@ccrisan please update this page also https://github.com/ccrisan/motioneye/wiki/Install-On-Ubuntu

ccrisan, 2017-05-02:
Does ubuntu default to python3 these days?

RobertLHarris, 2018-09-19:
Probably, but more important.  

{0}:/var/log>head /usr/local/bin/meyectl                                                                                                                    **#!/usr/bin/python3**                                                                                                                                                                                                                                                                                                      # -*- coding: utf-8 -*-                                                                                                                                     import re                                                                                                                                                   import sys

MyFriendlyUserName, 2019-06-21:
Please update the documentation for Raspbian (at least for Stretch (9.9 )) as well. Thank you!

bragofsky, 2020-08-28:
Same issue in ubuntu 20.04, 

ago 28 00:00:34 brabuntu systemd[1]: Started motionEye Server.
ago 28 00:00:35 brabuntu meyectl[984]: Traceback (most recent call last):
ago 28 00:00:35 brabuntu meyectl[984]:   File "/usr/local/bin/meyectl", line 5, in <module>
ago 28 00:00:35 brabuntu meyectl[984]:     from motioneye.meyectl import main
ago 28 00:00:35 brabuntu meyectl[984]:   File "/usr/local/lib/python3.8/dist-packages/motioneye/meyectl.py", line 28, in <module>
ago 28 00:00:35 brabuntu meyectl[984]:     import settings
ago 28 00:00:35 brabuntu meyectl[984]: ModuleNotFoundError: No module named 'settings'
ago 28 00:00:35 brabuntu systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
ago 28 00:00:35 brabuntu systemd[1]: motioneye.service: Failed with result 'exit-code'.

tryed to install it with pip2 and got the folowing

sudo pip2 install motioneye
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Processing /root/.cache/pip/wheels/24/45/97/6cdec7430d1871c0a66171972c604033182ab104be32351e89/motioneye-0.42.1-py2-none-any.whl
Requirement already satisfied: jinja2 in /usr/local/lib/python2.7/dist-packages (from motioneye) (2.11.2)
Requirement already satisfied: pillow in /usr/lib/python2.7/dist-packages (from motioneye) (6.2.1)
Collecting pycurl
  Using cached pycurl-7.43.0.5.tar.gz (216 kB)
Requirement already satisfied: tornado<6,>=3.1 in /usr/local/lib/python2.7/dist-packages (from motioneye) (5.1.1)
Requirement already satisfied: MarkupSafe>=0.23 in /usr/local/lib/python2.7/dist-packages (from jinja2->motioneye) (1.1.1)
Requirement already satisfied: singledispatch in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (3.4.0.3)
Requirement already satisfied: futures in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (3.3.0)
Requirement already satisfied: backports-abc>=0.4 in /usr/local/lib/python2.7/dist-packages (from tornado<6,>=3.1->motioneye) (0.5)
Requirement already satisfied: six in /usr/local/lib/python2.7/dist-packages (from singledispatch->tornado<6,>=3.1->motioneye) (1.15.0)
Building wheels for collected packages: pycurl
  Building wheel for pycurl (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-h_WDRj/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-install-h_WDRj/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-05kHkH
       cwd: /tmp/pip-install-h_WDRj/pycurl/
  Complete output (20 lines):
  Using curl-config (libcurl 7.68.0)
  Using SSL library: OpenSSL/LibreSSL/BoringSSL
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-2.7
  creating build/lib.linux-x86_64-2.7/curl
  copying python/curl/__init__.py -> build/lib.linux-x86_64-2.7/curl
  running build_ext
  building 'pycurl' extension
  creating build/temp.linux-x86_64-2.7
  creating build/temp.linux-x86_64-2.7/src
  x86_64-linux-gnu-gcc -pthread -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-1x6jhf/python2.7-2.7.18~rc1=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DPYCURL_VERSION="7.43.0.5" -DHAVE_CURL_SSL=1 -DHAVE_CURL_OPENSSL=1 -DHAVE_CURL_SSL=1 -I/usr/include/python2.7 -c src/docstrings.c -o build/temp.linux-x86_64-2.7/src/docstrings.o
  In file included from src/docstrings.c:4:
  src/pycurl.h:5:10: fatal error: Python.h: No such file or directory
      5 | #include <Python.h>
        |          ^~~~~~~~~~
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  ----------------------------------------
  ERROR: Failed building wheel for pycurl
  Running setup.py clean for pycurl
Failed to build pycurl
DEPRECATION: Could not build wheels for pycurl which do not use PEP 517. pip will fall back to legacy 'setup.py install' for these. pip 21.0 will remove support for this functionality. A possible replacement is to fix the wheel build issue reported above. You can find discussion regarding this at https://github.com/pypa/pip/issues/8368.
Installing collected packages: pycurl, motioneye
    Running setup.py install for pycurl ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-h_WDRj/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-install-h_WDRj/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-D7vtjt/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python2.7/pycurl
         cwd: /tmp/pip-install-h_WDRj/pycurl/
    Complete output (20 lines):
    Using curl-config (libcurl 7.68.0)
    Using SSL library: OpenSSL/LibreSSL/BoringSSL
    running install
    running build
    running build_py
    creating build
[... cut ...]
ERROR: Command errored out with exit status 1: /usr/bin/python2 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-h_WDRj/pycurl/setup.py'"'"'; __file__='"'"'/tmp/pip-install-h_WDRj/pycurl/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-D7vtjt/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python2.7/pycurl Check the logs for full command output.
tatus=1/FAILURE
ago 28 00:00:35 brabuntu systemd[1]: motioneye.service: Failed with result 'exit-code'.