motioneye-project/motioneye #59: Motion graph

alexcashplay, 2015-11-26:
Is there any way to implement motion graph based on quantity of captured photos per hour? Same as its done in Zoneminder for example?

alexcashplay, 2015-12-01:
Hello? Should I wait for any update?

ccrisan, 2015-12-01:
Don't. This is definitely not a priority and there are many other pending requests. My apologies for not getting back to you with an answer sooner.

alexcashplay, 2015-12-01:
Bad do hear, thanks for the prompt answer. Will try implement myself then,

ccrisan, 2015-12-01:
Please do, if you have the necessary resources and knowledge. You could share it with us then (if you want, of course) and I'll consider integrating it into motionEye.

mvanbreeden, 2016-01-03:
Google UA would be nice, requires just a few lines of code, an extra account field in UI and you can track online in realtime. I'll see what I can do coming days.

eborned, 2016-01-14:
@mvanbreeden could you elaborate on the difficulties, just for future reference?

mvanbreeden, 2016-01-16:
The difficulties are threefold:
First of all the UA features are not quite as simpel as they seem. That
makes it difficult to grasp and use for end-users.
Secondly implementing UA feeds would mean to either pick and choose info
and categories to send upfront or to implement very broadly. Both are not
ideal for a project like motioneye.

Lastly: to keep motioneye light and compact, one would expect an elegant
and self-contained graph, UA would be shooting a fly with a bazooka.

My intentionaliteit is to experiment loosely with UA on other stuff first