motioneye-project/motioneye #60: wiki suggestion

sgiannulli, 2015-12-01:
Hi,
just a suggestion to update requirements on wiki page about how to compile mr-dave fork of motion:
"sudo apt-get install dh-autoreconf"
raspbian at least does not include it in standard image
Thanks

ccrisan, 2015-12-01:
Normally if you have followed the instructions from the [wiki page](https://github.com/ccrisan/motioneye/wiki/Compiling-Motion), on a Debian systems you would have installed `build-essential`, which should have normally pulled the `autoconf` package. If it didn't, just let me know and I'll explicitly add it to the wiki page.

sgiannulli, 2015-12-01:
I had build-essential already installed on my Raspbian and trying to compile mr-dave motion from a directory where the config script was missing the result was:
"autoreconf: command not found"
so I think autoconf is not pulled with build-essential package as reported in this page https://help.ubuntu.com/community/CompilingSoftware

ccrisan, 2015-12-01:
Alright then. I added autoconf to the packages that need to be installed in order to compile motion on Debian.

closed 2015-12-01 by ccrisan