motioneye-project/motioneye #62: Failed to start, to useful error message

guaycuru, 2015-12-02:
Hello,

I've followed https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian to install motioneye on my Raspberry B+, with the only exception that I've built motion from this repo: https://github.com/lowflyerUK/motion/tree/mmal-test.

For some reason, I can't start motioneye, and I can't even find an useful error message to try and determine what's going on:

```
sudo /etc/init.d/motioneye start
[....] Starting motioneye (via systemctl): motioneye.serviceJob for motioneye.service failed. See 'systemctl status motioneye.service' and 'journalctl -xn' for details.
 failed!
```

```/var/log/motioneye.log``` is always empty

```
$ sudo systemctl status motioneye.service
● motioneye.service - LSB: Start the motionEye server
   Loaded: loaded (/etc/init.d/motioneye)
   Active: failed (Result: exit-code) since Wed 2015-12-02 17:12:12 UTC; 2min 4s ago
  Process: 4085 ExecStart=/etc/init.d/motioneye start (code=exited, status=1/FAILURE)

Dec 02 17:12:12 guaypib motioneye[4085]: Starting motionEye server:Illegal instruction
Dec 02 17:12:12 guaypib motioneye[4085]: failed!
Dec 02 17:12:12 guaypib systemd[1]: motioneye.service: control process exited, code=exited status=1
Dec 02 17:12:12 guaypib systemd[1]: Failed to start LSB: Start the motionEye server.
Dec 02 17:12:12 guaypib systemd[1]: Unit motioneye.service entered failed state.
```

```
$ sudo journalctl -xn
-- Logs begin at Wed 2015-12-02 16:43:27 UTC, end at Wed 2015-12-02 17:14:37 UTC. --
Dec 02 17:12:12 guaypib systemd[1]: motioneye.service: control process exited, code=exited status=1
Dec 02 17:12:12 guaypib systemd[1]: Failed to start LSB: Start the motionEye server.
-- Subject: Unit motioneye.service has failed
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
--
-- Unit motioneye.service has failed.
--
-- The result is failed.
Dec 02 17:12:12 guaypib systemd[1]: Unit motioneye.service entered failed state.
Dec 02 17:12:12 guaypib sudo[4069]: pam_unix(sudo:session): session closed for user root
Dec 02 17:14:01 guaypib dhcpcd[588]: DHCPv6 packet too short from fe80::ae9e:17ff:fe65:46d8
Dec 02 17:14:16 guaypib sudo[4337]: pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/systemctl status motioneye.service
Dec 02 17:14:16 guaypib sudo[4337]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Dec 02 17:14:16 guaypib sudo[4337]: pam_unix(sudo:session): session closed for user root
Dec 02 17:14:37 guaypib sudo[4392]: pi : TTY=pts/0 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/journalctl -xn
Dec 02 17:14:37 guaypib sudo[4392]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
```

Any idea what could be wrong?

chrisf4lc0n, 2015-12-02:
Are you running Debian Jessie or Wheezy?

chrisf4lc0n, 2015-12-02:
Does it start through command:
sudo meyectl startserver -c /etc/motioneye/motioneye.conf
Does it create a motioneye.log in /var/log?

chrisf4lc0n, 2015-12-02:
I seem to be having the same problem now, strange thing is that I managed to get it running for a while and it went dead... Raspbian Jessie Minimal...
In the same time I am running another MotionEye on Ubuntu and have not got any problems!

The result after issuing journalctl -xn
root@DoorBell:/etc/systemd/system# journalctl -xn
-- Logs begin at Wed 2015-12-02 19:39:35 UTC, end at Wed 2015-12-02 21:18:14 UTC. --
Dec 02 21:17:08 DoorBell systemd[1]: Starting motionEye Server...
-- Subject: Unit motioneye.service has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit motioneye.service has begun starting up.
Dec 02 21:17:08 DoorBell systemd[1]: Started motionEye Server.
-- Subject: Unit motioneye.service has finished start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit motioneye.service has finished starting up.
-- 
-- The start-up result is done.
Dec 02 21:17:24 DoorBell systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
Dec 02 21:17:25 DoorBell systemd[1]: Unit motioneye.service entered failed state.
Dec 02 21:17:47 DoorBell systemd[1]: Starting motionEye Server...
-- Subject: Unit motioneye.service has begun with start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit motioneye.service has begun starting up.
Dec 02 21:17:48 DoorBell systemd[1]: Started motionEye Server.
-- Subject: Unit motioneye.service has finished start-up
-- Defined-By: systemd
-- Support: http://lists.freedesktop.org/mailman/listinfo/systemd-devel
-- 
-- Unit motioneye.service has finished starting up.
-- 
-- The start-up result is done.
Dec 02 21:18:04 DoorBell systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
Dec 02 21:18:04 DoorBell systemd[1]: Unit motioneye.service entered failed state.
Dec 02 21:18:14 DoorBell sudo[4956]: root : TTY=pts/0 ; PWD=/etc/systemd/system ; USER=root ; COMMAND=/bin/journalctl -xn
Dec 02 21:18:14 DoorBell sudo[4956]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
root@DoorBell:/etc/systemd/system# systemctl status motioneye.service
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; disabled)
   Active: failed (Result: signal) since Wed 2015-12-02 21:18:04 UTC; 30s ago
  Process: 4951 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=killed, signal=ILL)
 Main PID: 4951 (code=killed, signal=ILL)

Dec 02 21:17:48 DoorBell systemd[1]: Started motionEye Server.
Dec 02 21:18:04 DoorBell systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
Dec 02 21:18:04 DoorBell systemd[1]: Unit motioneye.service entered failed state.

Also:
root@DoorBell:/usr/bin# meyectl startserver -c /etc/motioneye/motioneye.conf
Illegal instruction

No motioneye.log created.

chrisf4lc0n, 2015-12-02:
root@DoorBell:/home/pi# systemctl status motioneye.service
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; disabled)
   Active: failed (Result: signal) since Wed 2015-12-02 21:54:54 UTC; 3s ago
  Process: 789 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=killed, signal=ILL)
 Main PID: 789 (code=killed, signal=ILL)

Dec 02 21:54:37 DoorBell systemd[1]: Started motionEye Server.
Dec 02 21:54:54 DoorBell systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
Dec 02 21:54:54 DoorBell systemd[1]: Unit motioneye.service entered failed state.

Cannot get my head around it :(

guaycuru, 2015-12-02:
@chrisf4lc0n 
I'm running Jessie Minimal.
```/var/log/motioneye.log``` is there, but it's always empty (0 bytes)
Running ```sudo meyectl startserver -c /etc/motioneye/motioneye.conf``` produces no output, but the service does not appear to be running.

ccrisan, 2015-12-03:
Do you guys use the precompiled motion daemon suggested [here](https://github.com/ccrisan/motioneye/wiki/motion-binaries/motion-mrdave-raspbian)? The SIGILL that you get (illegal instruction) makes me think that the precompiled binary is somehow not compatible with some versions of Raspbian. If yes, could you please remove it and try using the official version of motion?

chrisf4lc0n, 2015-12-03:
@guaycuru you should be using systemd script to get it started, not the init.d, as per instruction here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Debian Debian 8.

@ccrisan Trying what you mentioned about motion...

chrisf4lc0n, 2015-12-03:
OK, I installed motion from the official repos and managed to start it:
● motion.service - LSB: Start Motion detection
   Loaded: loaded (/etc/init.d/motion)
   Active: active (exited) since Thu 2015-12-03 09:12:37 UTC; 3min 30s ago

Dec 03 09:12:36 DoorBell motion[1411]: Not starting motion daemon, disabled via /etc/default/motion ... (warning).
Dec 03 09:12:37 DoorBell systemd[1]: Started LSB: Start Motion detection.
Dec 03 09:13:45 DoorBell systemd[1]: Started LSB: Start Motion detection.

But motioneye is still not running, spitting out the same error:
motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; disabled)
   Active: failed (Result: signal) since Thu 2015-12-03 09:15:05 UTC; 1min 57s ago
  Process: 1447 ExecStart=/usr/local/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=killed, signal=ILL)
 Main PID: 1447 (code=killed, signal=ILL)

Dec 03 09:14:51 DoorBell systemd[1]: Started motionEye Server.
Dec 03 09:15:05 DoorBell systemd[1]: motioneye.service: main process exited, code=killed, status=4/ILL
Dec 03 09:15:05 DoorBell systemd[1]: Unit motioneye.service entered failed state.

chrisf4lc0n, 2015-12-03:
Will try to compile it, but that on the Pi will take absolutely ages ;)

ccrisan, 2015-12-03:
@chrisf4lc0n don't bother compiling motion from sources on a P - just use the old stable version for now and I'll compile it for you, it that turns out to be the problem. Could you please instead run the following commands and let me know what the output is (if any):

    meyectl startserver -c /etc/motioneye/motioneye.conf -d

and

    motion -h

Also, what's the `motion_binary` set to in the `/etc/motioneye/motioneye.conf`?

chrisf4lc0n, 2015-12-03:
I have completely changed the approach and simply reverted back to Wheezy, I am still in process of installing it. Once everything is up and running I will report back. It might be a problem with Jessie...

chrisf4lc0n, 2015-12-03:
@ccrisan I am working on it. Long story short, I reinstalled the OS.

chrisf4lc0n, 2015-12-04:
OK I have got it... With the original motion installed from apt-get install it is absolutely fine:

pi@raspberrypi:~ $ sudo meyectl startserver -c /etc/motioneye/motioneye.conf -d
    INFO: hello! this is motionEye server 0.29
Illegal instruction
   DEBUG: reading main config from file /etc/motioneye/motion.conf...
   DEBUG: loading additional config structure for main, without separators
   DEBUG: additional config section: network
   DEBUG: listing config dir /etc/motioneye...
    INFO: cleanup started
    INFO: wsswitch started
    INFO: tasks started
    INFO: mjpg client garbage collector started
    INFO: server started
   DEBUG: running garbage collector for mjpg clients...
   DEBUG: running garbage collector for mjpg clients...
   DEBUG: loading additional config structure for main, with separators
   DEBUG: additional config section: network
   DEBUG: loading additional config structure for camera, with separators
   DEBUG: 304 GET / (192.168.0.5) 26735.20ms
   DEBUG: running garbage collector for mjpg clients...
   DEBUG: 200 GET /login/?_=1449215694396&_username=admin&_signature=5d130fd5fcd4465988d3d0f4ce5a43e7de41df24 (192.168.0.5) 35.95ms
   DEBUG: 200 GET /login/?_=1449215698749&_username=admin&_login=true&_signature=742d42efa95b0588f572cb43a69135e6b87f6972 (192.168.0.5) 59.47ms
   DEBUG: 200 POST /login/ (192.168.0.5) 75.49ms
   DEBUG: getting main config
   DEBUG: 200 GET /config/main/get/?_=1449215698947&_username=admin&_signature=376c86453ddd98cd3eaba97e0a738e9f588a73d3 (192.168.0.5) 99.61ms
   DEBUG: listing cameras
   DEBUG: 200 GET /config/list/?_=1449215699741&_username=admin&_signature=7c813c7d2ea797e6e69357cbe94ebdf8e5870ca7 (192.168.0.5) 111.45ms
   DEBUG: listing cameras
   DEBUG: listing v4l2 devices...
   DEBUG: 200 GET /config/list/?_=1449215703467&username=&password=&proto=v4l2&_username=admin&_signature=3b3ef79f862d4df1cbd6535274e18bab9a9583f0 (192.168.0.5) 325.59ms
   DEBUG: running garbage collector for mjpg clients...
   DEBUG: running cleanup process...
   DEBUG: cleaning up pictures...
   DEBUG: cleaning up movies...
   DEBUG: cleanup done
   DEBUG: running garbage collector for mjpg clients...
   DEBUG: running garbage collector for mjpg clients...
   DEBUG: running garbage collector for mjpg clients...

Now however I seemed to encounter another problem. It does not detect my Raspi Camera Module, which is strange, as when I installed MotionEyeOS it picked it up automatically.
I have added bcm2835-v4l2 to /etc/modules as well as start_x=1 to /boot/config.txt but still nothing...

ccrisan, 2015-12-04:
@chrisf4lc0n there's still an *Illegal Instruction* in your log so nope, things are not alright. You didn't answer all my questions. What does `motion -h` say and what's the `motion_binary` set to in `/etc/motioneye/motioneye.conf`?

chrisf4lc0n, 2015-12-04:
OK I got to the bottom of it:
1. MotionEye starts if Raspi Camera Module is not configured.
2. When I load sudo bcm2835-v4l2 and add camera through the WebUI regardless what motion binary do I use it spits out the errors shown above and crashes the MotionEye.

ccrisan, 2015-12-04:
Let's try again: what does `motion -h` say and what's the `motion_binary` set to in `/etc/motioneye/motioneye.conf`?

chrisf4lc0n, 2015-12-04:
No luck with motion-mmal:

Exception: motion failed to start
   ERROR: 500 POST /config/add/?_=1449220263265&_username=admin&_signature=b7a5acbcddcfb79d9892aeb9996c55afe8f3e32f (192.168.0.5) 2718.80ms
   ERROR: motion not running, starting it
   DEBUG: starting motion
   DEBUG: using motion binary "//home/pi/motion-mmal-from-cctvpi/motion"
   ERROR: failed to start motion: motion failed to start
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 325, in checker
    motionctl.start()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 122, in start
    raise Exception('motion failed to start')
Exception: motion failed to start
   DEBUG: running garbage collector for mjpg clients...
   DEBUG: running cleanup process...
   DEBUG: cleaning up pictures...
   DEBUG: cleaning up movies...
   DEBUG: cleanup done
   ERROR: motion not running, starting it
   DEBUG: starting motion
   DEBUG: using motion binary "//home/pi/motion-mmal-from-cctvpi/motion"
   ERROR: failed to start motion: motion failed to start
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 325, in checker
    motionctl.start()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 122, in start
    raise Exception('motion failed to start')
Exception: motion failed to start
   DEBUG: running garbage collector for mjpg clients...

Can you provide motion binary from MotionEyeOS, as that one was the only one which seemed to be working?

chrisf4lc0n, 2015-12-04:
pi@raspberrypi:~/motion-mmal-from-cctvpi $ sudo motion -h
pi@raspberrypi:~/motion-mmal-from-cctvpi $ 

If it comes to motion_binary I must have tried everything...
1. DEBUG: using motion binary "/usr/local/bin/motion"
2. DEBUG: using motion binary "/usr/bin/motion"
3. DEBUG: using motion binary "/home/pi/motion-mmal-from-cctvpi/motion"

ccrisan, 2015-12-04:
So `motion -h` shows now output? That's odd. Let's see what motion versions you have installed across your system:

    `/usr/local/bin/motion -h`

    `/usr/bin/motion -h`

    `/home/pi/motion-mmal-from-cctvpi/motion -h`

chrisf4lc0n, 2015-12-04:
pi@raspberrypi:/etc/motioneye $ sudo /usr/local/bin/motion -h
pi@raspberrypi:/etc/motioneye $

pi@raspberrypi:/etc/motioneye $ sudo /usr/bin/motion -h
pi@raspberrypi:/etc/motioneye $

pi@raspberrypi:/etc/motioneye $ sudo /home/pi/motion-mmal-from-cctvpi/motion -h
/home/pi/motion-mmal-from-cctvpi/motion: error while loading shared libraries: libavformat.so.53: cannot open shared object file: No such file or directory

chrisf4lc0n, 2015-12-04:
Just to mention /usr/local/bin/motion is the one provided in MotionEye wiki precompiled for Raspbian.

ccrisan, 2015-12-04:
I would expect `/usr/bin/motion` to behave, since it's the officially provided one (or is it?). Could you please reinstall motion from the repos so that we can know for sure `/usr/bin/motion` is the 3.2.12 version that won't work on Raspbian? Then I can start digging in and come up with a fix as well as a new precompiled version.

chrisf4lc0n, 2015-12-04:
I must have reinstalled it about 3 times already, but here goes nothing...
This time I will remove all the other versions I have got installed as well as I purge the repo installation...
I assume that for the MotionEyeOS you are using Arch Linux, so that motion version would be no good to us?

ccrisan, 2015-12-04:
MotionEyeOS is based on buildroot and every release is tested by me before releasing it. It has nothing to do with Arch or Raspbian.

chrisf4lc0n, 2015-12-04:
Removal:
pi@raspberrypi:~ $ sudo apt-get purge motion
Reading package lists... Done
Building dependency tree        
Reading state information... Done
The following package was automatically installed and is no longer required:
  libjpeg8
Use 'apt-get autoremove' to remove it.
The following packages will be REMOVED:
  motion*
0 upgraded, 0 newly installed, 1 to remove and 1 not upgraded.
After this operation, 746 kB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 34920 files and directories currently installed.)
Removing motion (3.2.12+git20140228-4+b2) ...
Purging configuration files for motion (3.2.12+git20140228-4+b2) ...
userdel: user 'motion' does not exist
Processing triggers for man-db (2.7.0.2-5) ...
pi@raspberrypi:~ $ sudo apt-get autoremove 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  libjpeg8
0 upgraded, 0 newly installed, 1 to remove and 1 not upgraded.
After this operation, 230 kB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 34892 files and directories currently installed.)
Removing libjpeg8:armhf (8d1-2) ...
pi@raspberrypi:~ $ sudo rm /usr/local/bin/motion
pi@raspberrypi:~ $ sudo rm /usr/bin/motion
rm: cannot remove ‘/usr/bin/motion’: No such file or directory
pi@raspberrypi:~ $ ls
433Utils  motion-mmal-from-cctvpi  motion-mrdave-raspbian  wiringPi
pi@raspberrypi:~ $ sudo rm -R motion-mmal-from-cctvpi/
pi@raspberrypi:~ $ sudo rm motion-mrdave-raspbian 
pi@raspberrypi:~ $ 

Installation:
pi@raspberrypi:~ $ sudo apt-get install motion
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  libjpeg8
Suggested packages:
  mysql-client postgresql-client
The following NEW packages will be installed:
  libjpeg8 motion
0 upgraded, 2 newly installed, 0 to remove and 1 not upgraded.
Need to get 0 B/338 kB of archives.
After this operation, 977 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Preconfiguring packages ...
Selecting previously unselected package libjpeg8:armhf.
(Reading database ... 34884 files and directories currently installed.)
Preparing to unpack .../libjpeg8_8d1-2_armhf.deb ...
Unpacking libjpeg8:armhf (8d1-2) ...
Selecting previously unselected package motion.
Preparing to unpack .../motion_3.2.12+git20140228-4+b2_armhf.deb ...
Unpacking motion (3.2.12+git20140228-4+b2) ...
Processing triggers for man-db (2.7.0.2-5) ...
Processing triggers for systemd (215-17+deb8u2) ...
Setting up libjpeg8:armhf (8d1-2) ...
Setting up motion (3.2.12+git20140228-4+b2) ...
Adding group `motion' (GID 113) ...
Done.
Warning: The home dir /var/lib/motion you specified already exists.
Adding system user `motion' (UID 109) ...
Adding new user `motion' (UID 109) with group `motion' ...
The home directory `/var/lib/motion' already exists.  Not copying from `/etc/skel'.
adduser: Warning: The home directory `/var/lib/motion' does not belong to the user you are currently creating.
Adding user `motion' to group `video' ...
Adding user motion to group video
Done.
Processing triggers for libc-bin (2.19-18+deb8u1) ...
Processing triggers for systemd (215-17+deb8u2) ...
pi@raspberrypi:~ $ cd /var/lib/motion
pi@raspberrypi:/var/lib/motion $ ls
pi@raspberrypi:/var/lib/motion

Testing:
pi@raspberrypi:/var/lib/motion $ sudo motion -h
pi@raspberrypi:/var/lib/motion $

chrisf4lc0n, 2015-12-04:
Dead as a Dodo :(

ccrisan, 2015-12-04:
Then it's an upstream problem. I'll look into it during this weekend.

chrisf4lc0n, 2015-12-04:
You are the best mate!

chrisf4lc0n, 2015-12-04:
Does compiling motion on Pi would solve the problem?

ccrisan, 2015-12-04:
I couldn't tell before investigating what's actually going on.

chrisf4lc0n, 2015-12-04:
I will start compiling motion on Pi and will report back...

chrisf4lc0n, 2015-12-04:
Compiling Mr Dave motion from source did not do a thing:

pi@raspberrypi:~/motion $ sudo ./configure --prefix=/usr --with-ffmpeg=/usr/lib/arm-linux-gnueabihf  --with-ffmpeg-headers=/usr --without-pgsql --without-sdl --without-sqlite3 --without-mysql
configure: WARNING: unrecognized options: --with-ffmpeg-headers
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking minix/config.h usability... no
checking minix/config.h presence... no
checking for minix/config.h... no
checking whether it is safe to define __EXTENSIONS__... yes
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking for Darwin... no
checking for *BSD... no
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking threads... yes
checking for sdl... skipped
checking for libjpeg-turbo... skipping
checking for libjpeg-mmx... skipping
checking for jpeg_set_defaults in -ljpeg... yes
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking for FFMPEG... yes
checking for sqlite3... skipping
checking for mysql support... skipped
checking for PostgreSQL... skipped
checking for ANSI C header files... (cached) yes
checking stdio.h usability... yes
checking stdio.h presence... yes
checking for stdio.h... yes
checking for unistd.h... (cached) yes
checking for stdint.h... (cached) yes
checking fcntl.h usability... yes
checking fcntl.h presence... yes
checking for fcntl.h... yes
checking time.h usability... yes
checking time.h presence... yes
checking for time.h... yes
checking signal.h usability... yes
checking signal.h presence... yes
checking for signal.h... yes
checking sys/ioctl.h usability... yes
checking sys/ioctl.h presence... yes
checking for sys/ioctl.h... yes
checking sys/mman.h usability... yes
[... 118 lines cut ...]
mkdir -p /usr/share/man/man1
mkdir -p /usr/etc/motion
mkdir -p /usr/share/doc/motion 
mkdir -p /usr/share/motion/examples
install -m 644 motion.1 /usr/share/man/man1
install -m 644 CHANGELOG COPYING CREDITS README motion_guide.html /usr/share/doc/motion 
install -m 644 *.conf /usr/share/motion/examples
install motion.init-Debian motion.init-Fedora motion.init-FreeBSD.sh /usr/share/motion/examples
install -m 644 motion-dist.conf /usr/etc/motion
install -m 644 thread1-dist.conf /usr/etc/motion
install -m 644 thread2-dist.conf /usr/etc/motion
install -m 644 thread3-dist.conf /usr/etc/motion
install -m 644 thread4-dist.conf /usr/etc/motion
for prog in motion; \
do \
(install $prog /usr/bin ); \
done
--------------------------------------------------------------------------------
Install complete! The default configuration file, motion-dist.conf, has been
installed to /usr/etc/motion. You need to rename/copy it to motion.conf
for Motion to find it. More configuration examples as well as init scripts
can be found in /usr/share/motion/examples.

pi@raspberrypi:~/motion $ sudo motion -h
pi@raspberrypi:~/motion $

chrisf4lc0n, 2015-12-04:
Why does it capitalise some the lines?

chrisf4lc0n, 2015-12-04:
I finally got something more than an empty output:
root@DoorBell:/home/pi/motion# motion -h
Illegal instruction

ccrisan, 2015-12-04:
Eh well it's ffmpeg. It's ffmpeg that's been killing me ever since I started working on motionEye. Somehow the ffmpeg version offered by the 3rd party deb-multimedia repo crashes with Illegal Instruction (and so does any version of motion after installing this ffmpeg). The official repos don't offer ffmpeg at all, thanks to Debian and their issue with ffmpeg developers (or something of that sort). I've got no solution for this.

I'm trying to compile ffmpeg from sources, but apart from the fact that it takes ages, I wouldn't know how to make a deb out of it.

chrisf4lc0n, 2015-12-05:
At least you have identified the problem. I will have butchers on the ffmpeg myself and report back, if I will not succeed I simply either move to Arch or install Wheezy back.
Thanks a lot!

chrisf4lc0n, 2015-12-05:
3h+ and it ffmpeg compile is still going! I will not give up so easily! I will create deb package, test it and provide to you if it is working...

chrisf4lc0n, 2015-12-06:
Compiling the ffmpeg is not the thing you really want to do on the Pi! I left it for the night after over 3h in...
It finished, but with at least a dozen of errors/warnings, so some modifications in ./configure would be needed or it could be missing some libs.
I am afraid I have to admit that I have failed..., sorry!
Having to try to compile it once again would be like going through hell...
I am open to any suggestions.
At the moment I created a simple RTSP stream from my DoorBell Pi to my main server and everything gets handled on the Ubuntu 14.04.3 Server with MotionEye.

ccrisan, 2015-12-06:
I got it working and I am in the process of updating the wiki with ffmpeg-related instructions. Stay tuned!

ccrisan, 2015-12-06:
@chrisf4lc0n check out the first step here: https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian and let me know if this fixes your problem.

chrisf4lc0n, 2015-12-06:
@ccrisan I can confirm motion is working with no problems whatsoever!
You are the best Calin!

pi@DoorBell:~ $ sudo motion -h
motion Version 3.2.12+git20140228, Copyright 2000-2005 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen

usage:	motion [options]

Possible options:

-n			Run in non-daemon mode.
-s			Run in setup mode.
-c config		Full path and filename of config file.
-d level		Log level (1-9) (EMR, ALR, CRT, ERR, WRN, NTC, ERR, DBG, ALL). default: 6 / NTC.
-k type			Type of log (COR, STR, ENC, NET, DBL, EVT, TRK, VID, ALL). default: ALL.
-p process_id_file	Full path and filename of process id file (pid file).
-l log file 		Full path and filename of log file.
-m			Disable motion detection at startup.
-h			Show this screen.

Motion is configured using a config file only. If none is supplied,
it will read motion.conf from current directory, ~/.motion or /etc/motion.

ccrisan, 2015-12-06:
That's great! I'll close the issue then.

closed 2015-12-06 by ccrisan

ouafnico, 2016-05-24:
Hi all

I'm trying to install motioneye on RPi, but even following the wiki for raspbian, it's not working.

# motion
Instruction non permise

I have tried other version of motion, ffmpeg, but same problem...