motioneye-project/motioneye #63: Not detecting motion

TheMightyInferno, 2015-12-02:
I have motioneye setup on my Raspberry pi 2 and at first everything was working fine, but now it has completely stopped detecting any motion. No settings or anything has been changed since setup. Any ideas?

chrisf4lc0n, 2015-12-02:
Can you debug and post it here when there is motion in front of the cameras?

ccrisan, 2015-12-03:
Please:
1) restart your PI and see if the problem goes away - then we can know it's a motion/motionEye problem that needs further investigation.
2) If it doesn't, please send me your config files (motion.conf, thread-\*.conf), *making sure to remove any  sensitive data (read: passwords) from them*.

ccrisan, 2015-12-26:
This looks like an abandoned issue to me so I'll close it for now.

closed 2015-12-26 by ccrisan

italianjoe69, 2017-03-07:
@ccrisan - Great work on motionEye!!!  

I am having this same issue on 2 systems.  Both are Raspberry Pi3 running motionEyeos with Logitech cameras.  The motion was working fine when I first built the systems.  I then went to edit the motion mask and now the frame change sits at 0 and nothing is triggered.  I tried disabling the mask and put all settings to allow anything to be picked up as motion and no luck.  Rebooted the systems many times and no luck.

motionEye Version	20170212
Motion Version	4.0.1+git37b3595

[thread-1.conf.txt](https://github.com/ccrisan/motioneye/files/824619/thread-1.conf.txt)
[motion.conf.txt](https://github.com/ccrisan/motioneye/files/824608/motion.conf.txt)

TIA

metalshreds, 2017-06-15:
@italianjoe69 Do you have the **frame change threshold** to 0% ? I was having the same problem and just increasing it by any amount made motion detection work again with the mask turned on.