motioneye-project/motioneye #66: Email

Sean2603, 2015-12-08:
Ive tried editing the sendmail.py to make the email a "nicer" format but the changes never seem to take effect?

Am i editing the right file?
Also would you consider adding support for sms alerts?

ccrisan, 2015-12-08:
`sendmail.py` is indeed the file that needs to be changed. More precisely, the two vars at the beginning called `messages` and `subjects` are to be modified per your needs. Have you restarted motionEye after making those changes?

Nope, SMS support is not planned.

chrisf4lc0n, 2015-12-09:
If it comes to SMS you can do it yourself, just Google mail to SMS, then setup a separate notification and in "Run Command" option set it all up.

closed 2015-12-26 by ccrisan