motioneye-project/motioneye #67: MotionEye camera view doesn't work when motion auth is enabled

guaycuru, 2015-12-08:
Everything works fine until I enable authentication for motion (stream_auth_method 1).
Whenever I do that, the camera live view stops working on the motioneye web ui. Why is that?
I've added the camera as a local camera (Raspberry Cam) so I do not understand why that happens. Is there a way to have auth enabled and the live view working?

ccrisan, 2015-12-08:
What version of motionEye are you using?

guaycuru, 2015-12-08:
@ccrisan 
```
$ pip show motioneye
---
Name: motioneye
Version: 0.29
```

ccrisan, 2015-12-09:
It's a bug indeed, a but that has been resurrected recently. I'll fix it asap.

guaycuru, 2015-12-09:
Great, thanks @ccrisan !

ccrisan, 2015-12-12:
The bug has been fixed and the fix will be available with the soon-to-be-released 0.29.1.

closed 2015-12-12 by ccrisan

guaycuru, 2015-12-21:
@ccrisan just tested on 0.29.1 and got the same result. :(

ccrisan, 2015-12-22:
Have you restarted motionEye server after upgrading? Does the UI indicate version 0.29.1? If yes, could you please try readding your cameras?

guaycuru, 2015-12-22:
Yes, I have restarted the motionEye server and the raspberry pi just in case.
The UI does indicate version 0.29.1: ```Current Version	0.29.1```
I'm not sure what you mean by "please try readding your cameras".

ccrisan, 2015-12-22:
Can you remove the camera and add it back again?

guaycuru, 2015-12-24:
@ccrisan removing and re adding the camera worked! Thanks a lot!