motioneye-project/motioneye #69: Send email only from specified cameras

chrisf4lc0n, 2015-12-09:
I am not sure if this is a bug or this has not been implemented yet. I want MotionEye to send me email notifications only from 1 camera, but even I set it up just for 1 camera it sends me emails when it detects the movement on undefined cameras too.

ccrisan, 2015-12-12:
Are you sure about this behaviour? Normally each camera has its own email settings so they shouldn't interfere. Can I see yout `motion.conf` and your `thread-*.conf` files? Please replace passwords with xxxxxx or so.

chrisf4lc0n, 2015-12-12:
motion.conf:
# @enabled on
# @show_advanced on
# @normal_password xxx
# @admin_password xxx
# @admin_username admin
# @normal_username user

control_port 7999

control_html_output on
control_localhost on

webcontrol_html_output on
webcontrol_port 7999
webcontrol_localhost on

thread thread-2.conf
thread thread-1.conf

thread-1.conf:
# @webcam_resolution 100
# @upload_subfolders on
# @upload_server 
# @enabled on
# @network_server 
# @upload_username 
# @motion_detection on
# @upload_port 
# @upload_location 
# @preserve_movies 30
# @network_username 
# @id 1
# @upload_password 
# @upload_method post
# @working_schedule_type outside
# @webcam_server_resize off
# @network_password 
# @upload_service gdrive
# @name Boiler Room
# @preserve_pictures 1
# @storage_device custom-path
# @network_share_name 
# @upload_enabled off
# @working_schedule 

ffmpeg_output_movies on
stream_quality 85
threshold 15051
quality 50
noise_level 31
pre_capture 20
noise_tune on
stream_maxrate 30
output_pictures on
stream_localhost on
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
text_changes off
movie_filename %Y-%m-%d/%H-%M-%S
auto_brightness on
stream_port 8081
rotate 0
stream_auth_method 0
lightswitch 0
framerate 30
emulate_motion off
snapshot_filename 
snapshot_interval 0
minimum_motion_frames 10
stream_motion off
target_dir /mnt/CCTV/Boiler_Room
text_double on
post_capture 20
stream_authentication user:xxx
on_picture_save /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf -l picture_save %t %f
on_movie_end /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf -l movie_end %t %f
text_left Boiler Room
picture_filename %Y-%m-%d/%H-%M-%S
locate_motion_style redbox
locate_motion_mode off
netcam_keepalive on
max_movie_time 0
on_event_end /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf -l stop %t
text_right %Y-%m-%d\n%T
on_event_start /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf -l start %t; /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc sendmail -c /etc/motioneye/motioneye.conf -l 'smtp.gmail.com' '587' 'email@gmail.com' 'xxx' 'True' 'email@gmail.com' 'motion_start' '%t' '%Y-%m-%dT%H:%M:%S' '5'
ffmpeg_bps 9216000
event_gap 10
netcam_tolerant_check on
netcam_url http://192.168.0.18/videostream.cgi?loginuse=admin&loginpas=xxx

thread-2.conf:
# @working_schedule_type outside
# @webcam_resolution 100
# @enabled on
# @name Front Door
# @network_share_name 
# @network_server 
# @motion_detection on
# @preserve_pictures 30
# @storage_device custom-path
# @preserve_movies 0
# @network_username 
# @network_password 
# @id 2
# @working_schedule 
# @webcam_server_resize off
# @upload_subfolders on
# @upload_server 
# @upload_username 
# @upload_port 
# @upload_location 
# @upload_password 
# @upload_method post
# @upload_service gdrive
# @upload_enabled off

stream_authentication user:xxx
snapshot_interval 0
ffmpeg_output_movies off
stream_motion off
target_dir /mnt/CCTV/Front_Door
netcam_url rtsp://192.168.0.23:554/ucast/11
text_double on
height 1080
netcam_userpass admin:xxx
stream_auth_method 0
width 1920
stream_quality 85
threshold 114048
quality 100
post_capture 10
noise_level 31
on_event_end /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf -l stop %t
text_left Front Door
pre_capture 10
noise_tune on
picture_filename 
locate_motion_style redbox
locate_motion_mode off
stream_maxrate 30
output_pictures off
netcam_keepalive off
stream_localhost on
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
text_changes off
netcam_tolerant_check on
event_gap 10
auto_brightness on
stream_port 8082
rotate 0
text_right 
on_event_start /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf -l start %t
ffmpeg_bps 645120
lightswitch 0
framerate 30
movie_filename %Y-%m-%d/%H-%M-%S
max_movie_time 0
emulate_motion off
minimum_motion_frames 10
snapshot_filename 

on_picture_save /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf -l picture_save %t %f
on_movie_end /usr/bin/python /usr/local/lib/python2.7/dist-packages/motioneye/meyectl.pyc relayevent -c /etc/motioneye/motioneye.conf -l movie_end %t %f

Yet I received email:
Motion has been detected by camera "Front Door/Ubunciak" at 2015-12-12 21:24:34 (local time).

motionEye.

chrisf4lc0n, 2015-12-13:
I have sent the config files to your email Calin. I am 100% sure it sends me email notifications from all cameras regardless what camera I choose to have notifications from.

ccrisan, 2015-12-13:
What motionEye version do you use? Can I see your `motion.log` and `motionEye.log` files after enabling debug logging and generating some motion so that emails are sent?

chrisf4lc0n, 2015-12-13:
I have just upgraded to 0.30rc1

motion.log:
 - flag 56
[2] [INF] [NET] netcam_image_conv: jpeg_error 0
[2] [INF] [NET] netcam_proc_jpeg: processing jpeg image - content length 35976
[1] [INF] [ENC] ffmpeg_avcodec_log: tcp_read_packet:
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: ret=1 c=24 [$]
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: id=0 len=1448
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: tcp_read_packet:
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: ret=1 c=24 [$]
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: id=0 len=826
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: tcp_read_packet:
 - flag 56
[2] [INF] [NET] netcam_check_content_type: Content-type image/jpeg
[2] [INF] [NET] netcam_check_content_length: Content-Length -1
[2] [INF] [NET] netcam_check_content_length: Content-Length 35936
[2] [INF] [NET] netcam_read_next_header: Found image header record
[2] [DBG] [NET] netcam_read_html_jpeg: Calculated frame time 75149.593750
[2] [INF] [NET] netcam_init_jpeg: ***new pic delay successful***
[2] [INF] [NET] netcam_init_jpeg: jpeg_error 0
[1] [INF] [ENC] ffmpeg_avcodec_log: ret=1 c=24 [$]
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: id=0 len=1448
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: tcp_read_packet:
 - flag 56
[2] [INF] [NET] netcam_image_conv: jpeg_error 0
[1] [INF] [ENC] ffmpeg_avcodec_log: ret=1 c=24 [$]
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: id=0 len=762
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: tcp_read_packet:
 - flag 56
[2] [INF] [NET] netcam_proc_jpeg: processing jpeg image - content length 35880
[1] [INF] [ENC] ffmpeg_avcodec_log: ret=1 c=24 [$]
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: id=0 len=1448
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: tcp_read_packet:
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: ret=1 c=24 [$]
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: id=0 len=731
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: tcp_read_packet:
 - flag 56
[2] [INF] [NET] netcam_check_content_type: Content-type image/jpeg
[2] [INF] [NET] netcam_check_content_length: Content-Length -1
[2] [INF] [NET] netcam_check_content_length: Content-Length 35868
[2] [INF] [NET] netcam_read_next_header: Found image header record
[2] [DBG] [NET] netcam_read_html_jpeg: Calculated frame time 75015.937500
[2] [INF] [NET] netcam_init_jpeg: ***new pic delay successful***
[2] [INF] [NET] netcam_init_jpeg: jpeg_error 0
[1] [INF] [ENC] ffmpeg_avcodec_log: ret=1 c=24 [$]
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: id=0 len=1448
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: tcp_read_packet:
 - flag 56
[1] [INF] [ENC] ffmpeg_avcodec_log: ret=1 c=24 [$]
 - flag 56
[2] [INF] [NET] netcam_image_conv: jpeg_error 0
[1] [INF] [ENC] ffmpeg_avcodec_log: id=0 len=689
[... 215 lines cut ...]
2015-12-13 17:31:08: [relayevent]    DEBUG: hello!
2015-12-13 17:31:08: [relayevent]    DEBUG: event = stop
2015-12-13 17:31:08: [relayevent]    DEBUG: thread_id = 2
2015-12-13 17:31:08: [relayevent]    DEBUG: reading main config from file /etc/motioneye/motion.conf...
2015-12-13 17:31:08: [motioneye]    DEBUG: recevied relayed event stop for thread id 2 (camera id 1)
2015-12-13 17:31:08: [motioneye]    DEBUG: clearing motion detected for camera with id 1
2015-12-13 17:31:08: [relayevent]    DEBUG: event successfully relayed
2015-12-13 17:31:08: [relayevent]    DEBUG: bye!
2015-12-13 17:31:08: [motioneye]    DEBUG: 200 POST /_relay_event/?_signature=283038ad42f2098fcdad21e53f04414f028b9533 (127.0.0.1) 4.45ms
2015-12-13 17:31:08: [relayevent]    DEBUG: hello!
2015-12-13 17:31:08: [relayevent]    DEBUG: event = movie_end
2015-12-13 17:31:08: [relayevent]    DEBUG: thread_id = 2
2015-12-13 17:31:08: [relayevent]    DEBUG: filename = /mnt/CCTV/Boiler_Room/2015-12-13/17-30-51.avi
2015-12-13 17:31:08: [relayevent]    DEBUG: reading main config from file /etc/motioneye/motion.conf...
2015-12-13 17:31:08: [motioneye]    DEBUG: recevied relayed event movie_end for thread id 2 (camera id 1)
2015-12-13 17:31:08: [motioneye]    DEBUG: adding task "make_movie_preview(/mnt/CCTV/Boiler_Room/2015-12-13/17-30-51.avi)" in 5 seconds
2015-12-13 17:31:08: [motioneye]    DEBUG: saving tasks to "/etc/motioneye/tasks.pickle"...
2015-12-13 17:31:08: [relayevent]    DEBUG: event successfully relayed
2015-12-13 17:31:08: [relayevent]    DEBUG: bye!
2015-12-13 17:31:08: [motioneye]    DEBUG: 200 POST /_relay_event/?_signature=38febf451819f5b6142924765275b0b23bbfe1d8 (127.0.0.1) 5.31ms
2015-12-13 17:31:09: [motioneye]    DEBUG: running garbage collector for mjpg clients...
2015-12-13 17:31:19: [motioneye]    DEBUG: executing task "make_movie_preview(/mnt/CCTV/Boiler_Room/2015-12-13/17-30-51.avi)"
2015-12-13 17:31:19: [motioneye]    DEBUG: saving tasks to "/etc/motioneye/tasks.pickle"...
2015-12-13 17:31:19: [motioneye]    DEBUG: running garbage collector for mjpg clients...
2015-12-13 17:31:19: [motioneye]    DEBUG: creating movie preview for /mnt/CCTV/Boiler_Room/2015-12-13/17-30-51.avi with an offset of 4 seconds...

ccrisan, 2015-12-13:
The log files appear to be incomplete. Please send me the entire files by email.

chrisf4lc0n, 2015-12-14:
I take it, that the 2 newest commits should fix my problem, but I am not sure how to update them files as when I issue upgrade command it either goes down to 0.29.1 or if I specify version it takes me to 0.30rc1. I can manually pull files and simply replace them, but I am sure that there is a bit more sophisticated way of doing that...

chrisf4lc0n, 2015-12-15:
OK, I have got it:
sudo pip install git+git://github.com/ccrisan/motioneye.git
does the job and pulls the newest commits, just compared the files...
I have not tested it yet, as I am making myself ready for work, but will do that once I am back :)
Thanks a ton m8!

ccrisan, 2015-12-15:
I didn't fix the mixed-up email notification bug yet. And yes, that's how you would install the very latest commit of motionEye, but I don't recommend that as that's very likely to be an unstable/experimental commit.

chrisf4lc0n, 2015-12-15:
Let me be a guinea pig..., I do not mind experimenting a bit and if anything comes out of it I will let you know, so you should know about it before the final release.

ccrisan, 2015-12-20:
@chrisf4lc0n thanks for being a *guniea pig* and for providing support for this project. Could you please try the latest commit and see if the problem with mixed-up emails has been fixed?

chrisf4lc0n, 2015-12-20:
On it...

chrisf4lc0n, 2015-12-20:
I can confirm that issue has now been resolved, so you can close it.
Thanks again!

ccrisan, 2015-12-20:
Thanks for testing it. And of course thanks for the beer. I'm off having that pint right now, so please keep that RTSP access on for a bit longer please :)

closed 2015-12-20 by ccrisan