motioneye-project/motioneye #70: Motioneye upgrade on CentOS

rafi1975, 2015-12-10:
Hi,

I've been trying to upgrade the existing motioneye to the latest version and getting the following errors:

/usr/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
Collecting motioneye
  Using cached motioneye-0.29.tar.gz
Requirement already up-to-date: tornado>=3.1 in /usr/lib64/python2.7/site-packages (from motioneye)
Requirement already up-to-date: jinja2 in /usr/lib/python2.7/site-packages (from motioneye)
Collecting pillow (from motioneye)
  Using cached Pillow-3.0.0.tar.gz
Collecting pycurl (from motioneye)
  Using cached pycurl-7.19.5.3.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 20, in <module>
      File "/tmp/pip-build-6tYKV1/pycurl/setup.py", line 731, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_source)
      File "/tmp/pip-build-6tYKV1/pycurl/setup.py", line 488, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/tmp/pip-build-6tYKV1/pycurl/setup.py", line 71, in __init__
        self.configure()
      File "/tmp/pip-build-6tYKV1/pycurl/setup.py", line 107, in configure_unix
        raise ConfigurationError(msg)
    __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-6tYKV1/pycurl

Any idea how to get it fixed?

Thanks,

Rafal

ccrisan, 2015-12-12:
On Debian the package providing the `curl-config` file is `libcurl4-openssl-dev`. Look for a similar package in CentOS.

ccrisan, 2015-12-26:
Did you manage to get it working?

rafi1975, 2015-12-28:
Loads of issues with importing. I lost yum when python was uninstalled.
Then got it back and did the pip install from scratch. That worked ok.

closed 2016-07-03 by ccrisan