motioneye-project/motioneye #71: some python errors

jeanmarc77, 2015-12-10:
Hello,

I ve updated to latest version (pip install motioneye --upgrade) and i can't start motioneye anymore : /

Thanks
[root@i3 ~]# systemctl status motioneye
● motioneye.service - motionEye Server
   Loaded: loaded (/etc/systemd/system/motioneye.service; enabled; vendor preset: disabled)
   Active: failed (Result: exit-code) since jeu 2015-12-10 14:35:07 CET; 1min 50s ago
  Process: 296 ExecStart=/usr/bin/meyectl startserver -c /etc/motioneye/motioneye.conf (code=exited, status=1/FAILURE)
 Main PID: 296 (code=exited, status=1/FAILURE)

déc 10 14:35:07 i3 meyectl[296]: File "/usr/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2380, in load
déc 10 14:35:07 i3 meyectl[296]: return self.resolve()
déc 10 14:35:07 i3 meyectl[296]: File "/usr/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2386, in resolve
déc 10 14:35:07 i3 meyectl[296]: module = __import__(self.module_name, fromlist=['__name__'], level=0)
déc 10 14:35:07 i3 meyectl[296]: File "/usr/lib/python3.5/site-packages/motioneye/meyectl.py", line 28, in <module>
déc 10 14:35:07 i3 meyectl[296]: import settings
déc 10 14:35:07 i3 meyectl[296]: ImportError: No module named 'settings'
déc 10 14:35:07 i3 systemd[1]: motioneye.service: Main process exited, code=exited, status=1/FAILURE
déc 10 14:35:07 i3 systemd[1]: motioneye.service: Unit entered failed state.
déc 10 14:35:07 i3 systemd[1]: motioneye.service: Failed with result 'exit-code'.

chrisf4lc0n, 2015-12-11:
I had the same on the Ubuntu, but that did not stop MotionEye from working properly.

ccrisan, 2015-12-11:
What do you get if you start motionEye manually: meyectl startserver -d -c /etc/motioneye/motioneye.conf?

jeanmarc77, 2015-12-11:
hi,
[root@i3 ~]# meyectl startserver -d -c /etc/motioneye/motioneye.conf
Traceback (most recent call last):
  File "/usr/bin/meyectl", line 9, in <module>
    load_entry_point('motioneye==0.29', 'console_scripts', 'meyectl')()
  File "/usr/lib/python3.5/site-packages/pkg_resources/__init__.py", line
568, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3.5/site-packages/pkg_resources/__init__.py", line
2720, in load_entry_point
    return ep.load()
  File "/usr/lib/python3.5/site-packages/pkg_resources/__init__.py", line
2380, in load
    return self.resolve()
  File "/usr/lib/python3.5/site-packages/pkg_resources/__init__.py", line
2386, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib/python3.5/site-packages/motioneye/meyectl.py", line 28, in
<module>
    import settings
ImportError: No module named 'settings'
[root@i3 ~]#

ccrisan, 2015-12-12:
@jeanmarc77 what version were you running before?

jeanmarc77, 2015-12-12:
Hello,

Well i dunno exactly ;) i use to install via "pip2 install motioneye".
It work just few days ago. (was last saturday if i remember correctly)

ccrisan, 2015-12-12:
motionEye is not compatible with python 3.x. Please use python 2.7.

closed 2015-12-12 by ccrisan

jeanmarc77, 2015-12-12:
Thanks, that was it ! sorry

resuser, 2016-12-20:
Ubuntu 14.04
How do we use python 2.7?  I have several versions of python installed (including 2.7) and I just broke my installation by updating.  I believe I fixed this a while back by modifying the upstart file, but I can't figure out how to get this to work now.  If motionEye requires 2.7, then why does it now explicitly ask for 2.7?

resuser, 2016-12-20:
I figured out what I did to get motionEye installed in the first place.

A few months ago, I installed 0.35.0.  After proper pip installation (following the wiki instructions), the error above occurs.  I discovered that this is due to a bug in the /usr/local/bin/meyectl file.  The first line needs to be changed to #!/usr/bin/python2.7

After that, everything worked... UNTIL I ran pip upgrade to go to 0.35.1.  The fix above does not seem to work on this version.  I had to go back to 0.35.0 and apply my fix again to get it to work.

Any ideas how to get the newest version to work?

To reproduce the problem on Ubuntu 14.04:
```
$ sudo pip install --upgrade motioneye
...
Successfully installed motioneye-0.35.1

$ sudo service motioneye restart
 * Restarting motionEye server...                                                                                       * Stopping motionEye server                                                                                    [ OK ] 
 * Starting motionEye server                                                                                           Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 9, in <module>
    load_entry_point('motioneye==0.35.1', 'console_scripts', 'meyectl')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 558, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2682, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2355, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2361, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python3.4/dist-packages/motioneye/meyectl.py", line 28, in <module>
    import settings
ImportError: No module named 'settings'
                                                                                                                [fail]
 * (failed)...        

$ sudo nano /usr/local/bin/meyectl
#apply fix that works in 0.35.0 - modified first line of /usr/local/bin/meyectl to use python2.7

$ sudo service motioneye restart 
 * Restarting motionEye server...                                                                                       * Stopping motionEye server                                                                                           No process in pidfile '/var/run/motioneye.pid' found running; none killed.
                                                                                                                [ OK ]
 * Starting motionEye server                                                                                           Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3080, in <module>
    @_call_aside
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3066, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 3093, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 653, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 666, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 839, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'motioneye==0.35.1' distribution was not found and is required by the application
                                                                                                                [fail]
 * (failed)...    

$ sudo pip install --upgrade motioneye==0.35.0
...
Successfully installed motioneye-0.35

$ sudo service motioneye restart 
 * Restarting motionEye server...                                                                                       * Stopping motionEye server                                                                                           No process in pidfile '/var/run/motioneye.pid' found running; none killed.
                                                                                                                [ OK ]
 * Starting motionEye server                                                                                           Traceback (most recent call last):
  File "/usr/local/bin/meyectl", line 9, in <module>
    load_entry_point('motioneye==0.35', 'console_scripts', 'meyectl')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 558, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2682, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2355, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2361, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
[... cut ...]
 * Restarting motionEye server...                                                                                       * Stopping motionEye server                                                                                           No process in pidfile '/var/run/motioneye.pid' found running; none killed.
                                                                                                                [ OK ]
 * Starting motionEye server                                                                                    [ OK ] 
```

ccrisan, 2016-12-21:
@resuser if you apply the "fix" to 0.35.1, do you get the same error?

resuser, 2016-12-21:
@ccrisan 
Please read through my code section above.  I have posted the output from both 0.35.0 and 0.35.1 with and without the "fix".  There are 3 traceback sections.  The first is 0.35.1 without the fix.  The second is 0.35.1 with the fix. The third is 0.35.0 without the fix.  At the end of my code section, you can see that when I apply the fix to 0.35.0 there is no traceback since motionEye starts working again.

Thanks

ccrisan, 2016-12-21:
My apologies, I read `#apply fix that works in 0.35.0 - modified first line of /usr/local/bin/meyectl to use python2.7` and I assumed you only tried that in 0.35.0. 

The root cause of your problem is that you have installed motionEye with pip for python3 but you're trying to run it with python2. The solution is to reinstall it with pip2 (pip for python2).

resuser, 2016-12-21:
Well I feel rather stupid now.  I thought I recalled pip2 not working when I first installed motionEye, which is why I was using pip, but it seems to be working fine now.  Sorry to trouble you, and thank you for the help.